intel-aero/meta-intel-aero #9: Including 32-bit multilib configuration for jam-stapl breaks numerous recipes in meta-ros

JochiPochi, 2016-10-06:
Several recipes for ROS packages from the Yocto layer https://github.com/bmwcarit/meta-ros experience packaging issues of the following kind: 

`
ERROR: genmsg-0.5.7-r0 do_package: QA Issue: genmsg: Files/directories were installed but not shipped in any package:
`
`
ERROR: roscpp-traits-0.5.7-r0 do_package: Fatal QA errors found, failing task.
`
`
ERROR: roscpp-traits-0.5.7-r0 do_package: Function failed: do_package
`
`
ERROR: genmsg-0.5.7-r0 do_package: Fatal QA errors found, failing task.
`
`
ERROR: genmsg-0.5.7-r0 do_package: Function failed: do_package
`
`
ERROR: roscpp-traits-0.5.7-r0 do_package: QA Issue: roscpp-traits: Files/directories were installed but not shipped in any package:
`
`
roscpp-traits: 3 installed and not shipped files. [installed-vs-shipped]
`

I have isolated the issue to come from https://github.com/intel-aero/meta-intel-aero/commit/f2374a86abe055d1b3cadf91ecbb7c40f1a8c275, specifically the additions to the `conf/local.conf` file to support a 32-bit arch cross-compilation scheme: https://github.com/intel-aero/meta-intel-aero/commit/f2374a86abe055d1b3cadf91ecbb7c40f1a8c275#diff-f95cc75ba253227ffb1e5b0c0705be2fR242

This prevents the a successful inclusion of ROS for the intel-aero platform.

intel-aero, 2016-10-06:
We are aware of this issue and it will be addressed in next release, which is due soon.

closed 2016-10-06 by intel-aero

JochiPochi, 2016-10-06:
Thanks!, I'll be looking forward to it.
What would you suggest to do on the mean time? 
Is it okay to remove the entire commit? (https://github.com/intel-aero/meta-intel-aero/commit/f2374a86abe055d1b3cadf91ecbb7c40f1a8c275)

omarrmz, 2016-10-06:
Hi, the jam binary is compiled statically anyway, these were wrongly kept from an alternate recipe that used actual sources, they are not needed. You can remove the following from the conf and if possible submit a patch (we have a patch to be included at the next release but will drop it if you submit one):

+require conf/multilib.conf
+MULTILIBS = "multilib:lib32"
+DEFAULTTUNE_virtclass-multilib-lib32 = "x86"

JochiPochi, 2016-10-07:
Awesome. I'll work on something, test and PR if it works well then. Thanks

intel-aero, 2016-10-12:
This issue is resolved in release v01.00.02