intel-aero/meta-intel-aero #13: VIDIOC_DQBUF error 19 when running librealsense examples

shening, 2016-12-02:
Issue Type: bug?

Summary: When I try to run any of the librealsense examples on the Aero board I get an error. I have updated both the bios and the OS to the latest

Description: 
in usr/share/librealsense/examples/cpp-multicam
Starting Intel RealSense R200 .... Done
terminate called after throwing an instance of 'std::runtime_error'
what(): VIDIOC_DQBUF error 19, No such device

Component affected: Realsense camera

BIOS Version: Exmaple: Aero-01.00.03

intel-aero, 2017-01-24:
Please update to latest [software release](https://github.com/intel-aero/meta-intel-aero/wiki/Upgrade-To-Latest-Software-Release) and try again

lucasdemarchi, 2017-04-17:
Should be solved in the latest release.

closed 2017-04-17 by lucasdemarchi