intel-aero/meta-intel-aero #14: Support for Ubuntu in Intel Aero

PeterQFR, 2017-01-20:
Issue Type: Feature/Task

Summary: Ubuntu has a mature package management system, which provides a large number of precompiled libraries such as boost, ros, opencv openpcl. Yocto does not.

Description: Can we get an Ubuntu 16.04 image for the Aero board 

Component affected: All features of the board

intel-aero, 2017-01-24:
Please send your request to the forum at https://communities.intel.com/community/tech/intel-aero

lucasdemarchi, 2017-03-28:
@PeterQFR agreed, but this is out of scope for this repository.

Btw, you can already install Ubuntu on Intel Aero. To get the full functionality of the HW you will need some patches here and there, particularly to the kernel. You could take a look on this repo to check what are the required patches and use this kernel for booting Ubuntu, Fedora, Arch or any other distro.

I'm closing this issue in order to keep track of what's currently being worked on or in near term plans.

closed 2017-03-28 by lucasdemarchi