intel-aero/meta-intel-aero #15: Provide the image SDK

mbelluzzo, 2017-01-26:
You already provide the pre-built yocto image ready to flash, why don't you provide the generated yocto sdk's for the image that can be generated by the same server+command line that created the image in first place. If you already do so, plz update the wiki + intel download center page accordingly.

It saves a lot of time and effort for developers who want to build apps on top of your product.

mbelluzzo, 2017-01-26:
```bitbake <whatever you need> -c populate_sdk```

whbruce, 2017-03-16:
+1. Should also provide the more powerful [Extensible SDK](https://wiki.yoctoproject.org/wiki/Extensible_SDK)

`bitbake intel-aero-image -c populate_sdk_ext`