intel-aero/meta-intel-aero #16: WIFI disconnection

ChristophTobler, 2017-01-26:
I had have some random disconnects when using the WIFI of the Aero. BIOS, OS and FPGA are all up to date.
```
BIOS_VERSION = Aero-01.00.12_Prod
OS_VERSION = v01.00.04
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xff 0xff
```
The dmesg output can be seen here: http://pasted.co/6e217e2b

vilvo, 2017-01-30:
It can be followed from kernel log line 752:
 `[   73.697332] iwlwifi 0000:01:00.0: Queue 4 stuck for 10000 ms.`
to `intel-aero/kernel-source/drivers/net/wireless/iwlwifi/pcie/tx.c` which dumps the log lines 753-793.
to `intel-aero/kernel-source/drivers/net/wireless/iwlwifi/pcie/rx.c` for line 794:
`[   74.117575] iwlwifi 0000:01:00.0: Hardware error detected.  Restarting.`
to `intel-aero/kernel-source/drivers/net/wireless/iwlwifi/pcie/trans.c:` for the lines 795-820
to `intel-aero/kernel-source/drivers/net/wireless/iwlwifi/iwl-io.c:` for the lines 821-830
then it starts `IWL Error Log Dump`, requests HW reset and returns to tx.c for line 885:
`[   86.224637] iwlwifi 0000:01:00.0: Failing on timeout while stopping DMA channel 8 [0xa5a5a5a2]`

How frequently you encounter these firmware issues?

ChristophTobler, 2017-01-31:
Hard to answer... every once in a while. They don't happen regularly, but randomly. Maybe every third time I connect

ChristophTobler, 2017-02-01:
To also have the information here:
I have a Dell XPS 13 with Ubuntu 16.04, Kernel: 4.10.0-994-generic and it has QCA6174 802.11ac Wireless Network Adapter
The disconnects are Aero specific and they happen irregularly. It does not happen with the normal network (at least not that I have noticed).

ChristophTobler, 2017-02-01:
We have the issue now on several devices and with two different Aeros. Here's another ```dmesg``` output http://pasted.co/6ac0907c

The devices:
* Dell XPS 13 with Ubuntu 16.04, Kernel: 4.10.0-994-generic and it has QCA6174 802.11ac Wireless Network Adapter
* Dell Alienware with Ubuntu 16.04, Kernel: 4.4.0-59-generic, WIFI: TP-LINK usb adapter and internal: Intel Wireless 3165
* HP Omen with Ubuntu 16.04, Kernel: 4.4.0-53-generic, WIFI: QCA6174 802.11ac Wireless Network Adapter

It can also happen that the WIFI is not visible at all and also (if visible) that no connection can be established.

vilvo, 2017-02-03:
Thanks @ChristophTobler for reporting. I tried to reproduce the issue with my bare Aero compute board using an automated cycle "power on - boot and grep dmesg for the error - power  off ". I ran it twice for 10 boots but could not reproduce the issue. There's probably a difference in our HW setup or this is HW specific. This was listed as a known issue in cold boot scenarios at https://github.com/intel-aero/meta-intel-aero/wiki/Known-Limitations-&-Work-Arounds so I definitely follow-up on this. Reliable WiFi is supports many key use cases.

lucasdemarchi, 2017-03-28:
Possible fix in #30.  After that's integrated, let's see if this improves.

lucasdemarchi, 2017-04-17:
I didn't have these disconnects anymore so closing this. We can re-open it again if needed.

closed 2017-04-17 by lucasdemarchi

RohanKurane, 2017-12-11:
I have noticed that the intel aero wifi Aero-xxxxx connection drops intermittently causing my application to fail. Why does this happen ? And is there a way to fix this ?

pumaking, 2019-05-28:
I'm having the same issue on an Intel Aero with the latest versions:

BIOS_VERSION = Aero-01.00.16
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.6.2 (pyro)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc2
AeroFC firmware version = unknown

This is very consistent and makes using Wifi for any application practically impossible.

I've attached some dmesg output: 
[aero-dmesg.txt](https://github.com/intel-aero/meta-intel-aero/files/3230088/aero-dmesg.txt)

Wifi seems to work okay for a few seconds (maybe ~ 5 to 10) after the dmesg output but then becomes extremely laggy (~ 2 - 10 seconds of latency). After some time of lagginess, the dmesg output will print and the cycle continues.

This is a problem both in Yocto (Linux 4.4.113-yocto-standard) and Ubuntu (4.4.76-...).

Is there a solution?

stevenjowens, 2019-07-23:
I'm seeing the same problem.  
[dmesg_07_18_2019.log](https://github.com/intel-aero/meta-intel-aero/files/3423572/dmesg_07_18_2019.log)

I can connect a laptop to the drone's hotspot and ping it about 1400-odd times before iwlwifi gives me "Hardware error detected. Restarting."

If I connect both the drone and the laptop to an external network it works fine, I let it keep pinging overnight and it never had the problem.

RohanKurane, 2019-07-23:
@pumaking  @stevenjowens 
How far is the UAV from your laptops ?
Also, is the UAV battery fully charged ? I've seen situations where if the battery is low, the hotspot either won't show or will drop like you both see.

stevenjowens, 2019-07-24:
The UAV is about 3 feet from my laptop and it's plugged into a power strip.

RohanKurane, 2019-07-25:
Does the same thing happen when you try outdoors ?

RohanKurane, 2019-07-26:
@stevenjowens - This is what I do. I have a USB adapter I connect to the laptop. I then connect to my home wifi or any other wifi (use a mifi if I am outdoors) as my primary wifi and then to the Aero hotspot using the USB adapter. That way I get reliable connectivity to the Aero hotspot. Try that and let me know how it goes for you.