intel-aero/meta-intel-aero #17: USB disconnects

ChristophTobler, 2017-01-30:
I also have problems getting a stable usb connection. I use ```sudo ifconfig <name> 192.168.7.1``` to set the IP and then I can connect with ```ssh root@192.168.7.2```. But usually after ~1min I get disconnected.

This is the ouput of ```dmesg``` on the aero:
```
[    6.888954] usb 2-4: new SuperSpeed USB device number 2 using xhci_hcd
[    6.940290] udevd[489]: failed to execute '/usr/local/bin/usb-R200-in_udev' '/usr/local/bin/usb-R200-in_udev': No such file or directory
[    6.967235] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[    6.970080] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[    6.971238] uvcvideo: Unable to create debugfs 2-2 directory.
[    6.972258] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[    6.973440] uvcvideo: Unable to create debugfs 2-2 directory.
[    6.973771] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.4/input/input5
[    6.975115] usbcore: registered new interface driver uvcvideo
[    6.975118] USB Video Class driver (1.1.1)
[    6.979028] NOHZ: local_softirq_pending 08
[    7.349047] NOHZ: local_softirq_pending 08
[    8.455382] NOHZ: local_softirq_pending 08
[   11.058927] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   11.059625] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   11.186927] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   11.187271] iwlwifi 0000:01:00.0: L1 Enabled - LTR Enabled
[   11.280887] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   11.304678] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   12.415367] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.415372] Bluetooth: BNEP filters: protocol multicast
[   12.415393] Bluetooth: BNEP socket layer initialized
[   12.419724] cherryview-pinctrl INT33FF:01: request() failed for pin 23
[   12.427044] cherryview-pinctrl INT33FF:01: pin-23 (INT33FF:01:414) status -16
[   13.369925] random: nonblocking pool is initialized
[   59.965088] serial8250: too much work for irq39
[   61.637765] serial8250: too much work for irq39
[   65.779958] serial8250: too much work for irq39
[   67.788440] serial8250: too much work for irq39
[   71.242480] serial8250: too much work for irq39
[   72.558296] serial8250: too much work for irq39
[   90.202296] serial8250: too much work for irq39
[   91.779111] serial8250: too much work for irq39
[   93.301081] serial8250: too much work for irq39
[  100.071983] serial8250: too much work for irq39
[  101.503165] serial8250: too much work for irq39
[  102.868234] serial8250: too much work for irq39
[  113.260915] serial8250: too much work for irq39
[  114.743131] serial8250: too much work for irq39
[  117.963209] serial8250: too much work for irq39
[  118.899672] serial8250: too much work for irq39
[  124.121901] serial8250: too much work for irq39
[  126.076408] serial8250: too much work for irq39
[  135.068071] serial8250: too much work for irq39
[  147.692916] serial8250: too much work for irq39
[  151.559838] serial8250: too much work for irq39
[  155.088738] serial8250: too much work for irq39
[  161.126032] serial8250: too much work for irq39
[  165.499261] serial8250: too much work for irq39
[  165.670239] serial8250: too much work for irq39
[  167.365762] serial8250: too much work for irq39
[  172.087731] g_ether gadget: high-speed config #1: CDC Ethernet (ECM)
[  172.621653] serial8250: too much work for irq39
[  178.257274] serial8250: too much work for irq39
[  181.301903] serial8250: too much work for irq39
[  188.029162] serial8250: too much work for irq39
[  202.056139] serial8250: too much work for irq39
[  206.889507] serial8250: too much work for irq39
[  206.941699] serial8250: too much work for irq39
[  207.558103] serial8250: too much work for irq39
[  212.268219] serial8250: too much work for irq39
[  212.740386] serial8250: too much work for irq39
[  223.867089] serial8250: too much work for irq39
[  226.630896] serial8250: too much work for irq39
[  229.477377] serial8250: too much work for irq39
[  229.556597] serial8250: too much work for irq39
[  231.961654] serial8250: too much work for irq39
[  232.462053] serial8250: too much work for irq39
[  234.578169] serial8250: too much work for irq39
[  239.800998] serial8250: too much work for irq39
[  242.054934] serial8250: too much work for irq39
[  250.933705] serial8250: too much work for irq39
[  276.998103] serial8250: too much work for irq39
[  277.111708] serial8250: too much work for irq39
[  284.885358] serial8250: too much work for irq39
```

and this is ```dmesg``` of my desktop:
```
[ 2279.421051] cdc_ether 1-9:1.0 usb0: register 'cdc_ether' at usb-0000:00:14.0-9, CDC Ethernet Device, f2:55:10:c7:eb:3a
[ 2279.427867] cdc_ether 1-9:1.0 enp0s20f0u9: renamed from usb0
[ 2279.452109] IPv6: ADDRCONF(NETDEV_UP): enp0s20f0u9: link is not ready
[ 2279.452250] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2279.452256] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2279.652318] userif-3: sent link down event.
[ 2279.652324] userif-3: sent link up event.
[ 2299.505039] userif-3: sent link down event.
[ 2299.505046] userif-3: sent link up event.
[ 2325.031929] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2325.031945] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2325.032360] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2325.060499] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2325.060894] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2325.235132] userif-3: sent link down event.
[ 2325.235136] userif-3: sent link up event.
[ 2370.025001] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2370.025009] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2370.025058] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2370.034206] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2370.034234] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2370.034238] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2387.489447] wlx98ded0115c14: authenticate with a0:c5:89:02:3e:15
[ 2387.509840] wlx98ded0115c14: send auth to a0:c5:89:02:3e:15 (try 1/3)
[ 2387.540019] wlx98ded0115c14: authenticated
[ 2387.543176] wlx98ded0115c14: associate with a0:c5:89:02:3e:15 (try 1/3)
[ 2387.551067] wlx98ded0115c14: RX AssocResp from a0:c5:89:02:3e:15 (capab=0x411 status=0 aid=1)
[ 2387.551822] wlx98ded0115c14: associated
[ 2387.566810] bridge-wlp4s0: disabling the bridge
[ 2387.583189] bridge-wlp4s0: down
[ 2387.583194] bridge-wlp4s0: detached
[ 2387.583217] /dev/vmnet: open called by PID 1636 (vmnet-bridge)
[ 2387.583221] /dev/vmnet: hub 0 does not exist, allocating memory.
[ 2387.583228] /dev/vmnet: port on hub 0 successfully opened
[ 2387.583232] bridge-wlx98ded0115c14: device is wireless, enabling SMAC
[ 2387.583233] bridge-wlx98ded0115c14: up
[ 2387.583275] bridge-wlx98ded0115c14: attached
[ 2387.766739] userif-3: sent link down event.
[ 2387.766741] userif-3: sent link up event.
[ 2415.044235] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2415.044252] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2415.044550] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2415.044560] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2415.062641] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2415.062695] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2415.062704] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2460.024246] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2460.024261] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2460.024544] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
[ 2460.024553] cdc_ether 1-9:1.0 enp0s20f0u9: kevent 12 may have been dropped
```

ChristophTobler, 2017-02-03:
@vilvo The baud rate is 1500k

lucasdemarchi, 2017-04-17:
I didn't have these connects anymore on the latest release, so closing this.

closed 2017-04-17 by lucasdemarchi