intel-aero/meta-intel-aero #18: Barometer offset

ChristophTobler, 2017-01-30:
http://logs.px4.io/plot_app?log=fea66d89-2473-4690-9508-c3d151be6543#Nav-Altitude-Estimate
The barometer shows an altitude of almost 6000 meters. Is it a driver issue?

ChristophTobler, 2017-02-01:
https://github.com/PX4/Firmware/pull/6485 does not seem to have changed anything. We see the same offset and the same noise
[image]

lfelipe, 2017-02-02:
@lucasdemarchi

lucasdemarchi, 2017-02-08:
@ChristophTobler  On my end I noticed it actually improved the noise a lot from +-1m to +=15cm (which is the expected for this sensor).

The absolute altitude is in fact a problem and I came to know this week that we can have different sensors on different boards. There's no way to differentiate MS5607 from MS5611 looking only to the registers. I'm going to submit a PR to use some heuristics in order to differentiate them.  This should remove the offset you are seeing.

ChristophTobler, 2017-02-08:
@lucasdemarchi Ok, good to know. Thanks!

ChristophTobler, 2017-02-09:
This issues will be tracked in https://github.com/PX4/Firmware/issues/6545

closed 2017-02-09 by ChristophTobler