intel-aero/meta-intel-aero #19: GPS jamming

ChristophTobler, 2017-01-30:
http://logs.px4.io/plot_app?log=fea66d89-2473-4690-9508-c3d151be6543#Nav-GPS-Noise-_-Jamming
The GPS jamming indicator is very high. This could be due to the FPGA interfering with the GPS. I guess this could be improved by isolating and grounding the FPGA.

vilvo, 2017-02-03:
Nice GPS jamming indicator graph, like a city skyline. On the more serious side, this would need an in-depth analysis. Do you have a scope? Please keep us updated if you look more into this.

ChristophTobler, 2017-02-03:
We will test the effects with the Typhoon and the Real Sense kit. There the board seems to be shielded. I wonder if that has similar issues/effects.

ChristophTobler, 2017-02-09:
This issues will be tracked in https://github.com/PX4/Firmware/issues/6546

closed 2017-02-09 by ChristophTobler