intel-aero/meta-intel-aero #21: Verify and Release Mavlink Router Updates

airmapinc, 2017-02-14:
The mavlink router service would not start after the first time. The service runs perfectly initially but does not run after reboot. This is not always the case as sometimes it does run after reboot. I can test as needed to make sure we run multiple times to assure functionality.

vilvo, 2017-02-14:
Thanks, adding localhost to mavlink router startup is clear but would you please pass the exact command for 

> python example mavlink client

in aero shell?

airmapinc, 2017-02-14:
I am adding a test file to check the mavlink router service. 

python mavrouter-test.py and look for the "Got Heartbeat..." message

vilvo, 2017-02-15:
Thanks, I had a chance to test this quickly with a compute board after hard reset...

<pre>
root@intel-aero:~# ps|grep mavlink
  570 root     13056 S    /usr/bin/mavlink-routerd -b 1500000 -e 192.168.7.255 -e 192.168.1.255 -e 127.0.0.1 /dev/ttyS1
root@intel-aero:~# python mavrouter-test.py 
Waiting for GPS Heartbeat...
</pre>
... but I'm afraid my compute board is flat-lining. 

On the serious side, I tested it with the RTF as well
<pre>
root@intel-aero:~# ps|grep mavlink
  597 root     13056 S    /usr/bin/mavlink-routerd -b 1500000 -e 192.168.7.255 -e 192.168.1.255 -e 127.0.0.1 /dev/ttyS1
  638 root      4408 S    grep mavlink
root@intel-aero:~# python mavrouter-test.py 
Waiting for GPS Heartbeat...
Got Heartbeat...
root@intel-aero:~#
</pre>

If I try to stop and start mavlink-routerd, all I get is never ending stream of 
<pre>
(../git/comm.cpp:374) Error sending udp packet (Network is unreachable)
(../git/comm.cpp:374) Error sending udp packet (Network is unreachable)
(../git/comm.cpp:374) Error sending udp packet (Network is unreachable)
(../git/comm.cpp:374) Error sending udp packet (Network is unreachable)
(../git/comm.cpp:374) Error sending udp packet (Network is unreachable)
</pre>
even when mavlink-routerd is running (pid 666 is just coincidence)
<pre>
root@intel-aero:~# ps|grep mavli
  666 root     13056 S    /usr/bin/mavlink-routerd -b 1500000 -e 192.168.7.255 -e 192.168.1.255 -e 127.0.0.1 /dev/ttyS1
</pre>
As I'm no mavlink expert, is this what you are seeing or are you seeing something else? Should we fix the mavlink version to a specific commit for debug instead of master - https://github.com/intel-aero/meta-intel-aero/blob/master/recipes-support/mavlink-router/mavlink-router_0.1.bb ?

airmapinc, 2017-02-15:
The Got Heartbeat... is what you are looking for. You can remove the -e 192.168.7.255 and -e 192.168.1.255 for testing purposes.

Test procedure:
>make sure mavlink_routerd is running and target is 127.0.0.1
>run python mavrouter-test.py
wait for Got Heartbeat...
pull power if you get a heartbeat and run test again. repeat a few times to make sure you get a heartbeat every time

Yes, we should mavlink version to a specific version on github. That is a good idea.

vilvo, 2017-02-17:
I ran the power cycling five times and "Got Heartbeat" every time. For this test I used [the latest released version](https://downloadcenter.intel.com/download/26500/UAV-installation-files-for-Intel-Aero-Platform). I agree that reliability issues should be solved. For that I appreciate all your and community ideas to reproduce the issue so it can be debugged further.

For the Aero fix release being planned - which specific version (sha1 or tag) of mavlink you'd like to have included?

lfelipe, 2017-03-29:
Being solved by #38 .

closed 2017-03-29 by lfelipe