intel-aero/meta-intel-aero #24: HSUART1(ttyS1) dropping bytes

zehortigoza, 2017-03-16:
Transmitting small chunks of bytes is fine at higher baudrates but when there is continuous transitions of data it start to drop some bytes.

On dmesg we can see several messages like this:
[ 224.450230] serial8250: too much work for irq39

A easy way to reproduce it is flash the FPGA with: /etc/aero_compute_board_only_fpga.jam connect a FTDI to the HSUART pins on the power connector and send continuous data from you host machine to Aero compute board.

This is the same HSUART used to communicate with AeroFC so is very important have a this HSUART working.

lucasdemarchi, 2017-03-28:
"too much work for irq39" is gone if we redirect the interrupts to another core or if we update the kernel to 4.10.  However there's still corruption on the received packets if we go beyond baudrate of 460800.  Investigation for proper fix is needed so moving assigning this as milestone M1, so we can take a look again on it.

mindThomas, 2017-05-04:
Please test with a baud rate of 921600 (see https://dev.px4.io/en/companion_computer/pixhawk_companion.html)

rohithnlr, 2017-07-18:
I flashed the FPGA with: /etc/aero_compute_board_only_fpga.jam connect a FTDI to the HSUART pins on the power connector 
send continuous data  from host system. when i checked cat/proc/tty/driver/serial on the aero board side rx counter was updating with framing errors.

lamping7, 2017-10-13:
@lucasdemarchi What's the status on this? Has faster baud rates been confirmed OK with the newest software release (v1.5.1)?

Also, for performance, would it be better to interface with `/dev/ttyS1` directly, rather than the mavlink router on the RTF kit?

zehortigoza, 2017-10-13:
@lamping7 Take a look at this https://github.com/intel-aero/meta-intel-aero/pull/272
but there still no official image release with this change in both sides(Aero compute board and PX4/Ardupilot).

If you don't need more than 1 application talking with the flight stack it would be better but if you need a ground station + anything else you will need MAVLink Router.

lucasdemarchi, 2017-10-13:
And mavlink-router makes it easy to save the flight stack logs.  As per resource consumption there's little reason not to use mavlink-router.

lamping7, 2017-10-13:
@zehortigoza OK, I changed the relevant files, but I am having issues getting this setup. Through QGC(v3.2.4) I cannot get the SYS_COMPANION parameter to persist. Every time I set it to 921600 and reboot, the AeroFC goes back to 460800 (luckly I have mavlink-router configured for both bauds). Before reboot, If I go to the shell console, and input `nsh> param show SYS_COMPANION` It shows 921600 and indicates that it is saved. Any idea what is going on?

lucasdemarchi, 2017-10-13:
@lamping7 it's forced unconditionally by the init script in PX4. See https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/4070_aerofc#L78

lamping7, 2017-10-13:
@lucasdemarchi That'll do it. Anyway to change this without having to make and flash new firmware?

lucasdemarchi, 2017-10-13:
Not that I know of.

lamping7, 2017-10-14:
@lucasdemarchi Whats the motivation for forcing this? Why not move it up to the autoconfig block to allow for users to set this in the future?

lamping7, 2017-10-16:
@lucasdemarchi I removed the param set from the init script and did some testing with 921600 baud rate. The results are mixed.

```
root@intel-aero:~# /etc/init.d/mavlink-routerd.sh status
● mavlink-router.service - MAVLink Router
   Loaded: loaded (/lib/systemd/system/mavlink-router.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2017-08-08 00:29:24 UTC; 7min ago
  Process: 311 ExecStartPre=/bin/bash -c echo 4 > /sys/class/tty/ttyS1/rx_trig_bytes (code=exited, status=0/SUCCESS)
 Main PID: 340 (mavlink-routerd)
   Memory: 17.9M
      CPU: 33.564s
   CGroup: /system.slice/mavlink-router.service
           └─340 /usr/bin/mavlink-routerd

Aug 08 00:29:24 intel-aero mavlink-routerd[340]: UART [7] speed = 921600
Aug 08 00:29:24 intel-aero mavlink-routerd[340]: Open TCP [9] 0.0.0.0:5760 *
Aug 08 00:29:24 intel-aero mavlink-routerd[340]: Baudrate 921600 responded, keeping it
Aug 08 00:29:25 intel-aero mavlink-routerd[340]: Logging target system_id=1 on 00034-2017-08-08_00-29-25.ulg
Aug 08 00:29:29 intel-aero mavlink-routerd[340]: 4 messages to unknown endpoints in the last 5 seconds
```

I'm getting just as many `serial8250: too much work for irq39` messages as I do with 460800, which is a few every two minutes or so when connected to the Aero with QGC, nothing else. 

Issues are present when running Mavros in a Docker container. (I won't paste the whole thing, but you'll get the point here.) 
```
[ WARN] [1502152348.350863535]: PR: request param #11 timeout, retries left 2, and 682 params still missing
[ WARN] [1502152349.112353739]: TM: Clock skew detected (0.018177572 s). Hard syncing clocks.
[ WARN] [1502152349.394121252]: PR: request param #12 timeout, retries left 2, and 681 params still missing
[ WARN] [1502152350.409939677]: PR: request param #13 timeout, retries left 2, and 680 params still missing
[ WARN] [1502152351.456561131]: PR: request param #14 timeout, retries left 2, and 679 params still missing
[ INFO] [1502152352.309826756]: WP: mission received
[ WARN] [1502152352.497921168]: PR: request param #15 timeout, retries left 2, and 678 params still missing
[ WARN] [1502152353.501116859]: PR: request param #16 timeout, retries left 2, and 677 params still missing
[ WARN] [1502152354.510080570]: PR: request param #17 timeout, retries left 2, and 676 params still missing
[ WARN] [1502152355.552402885]: PR: request param #18 timeout, retries left 2, and 675 params still missing
[ WARN] [1502152356.594861330]: PR: request param #19 timeout, retries left 2, and 674 params still missing
[ INFO] [1502152357.292144798]: HP: requesting home position
[ WARN] [1502152357.598670201]: PR: request param #20 timeout, retries left 2, and 673 params still missing
[ WARN] [1502152358.651716207]: PR: request param #21 timeout, retries left 2, and 672 params still missing
[ WARN] [1502152359.656920094]: PR: request param #22 timeout, retries left 2, and 671 params still missing
[ WARN] [1502152360.416255310]: TM: Clock skew detected (0.015289480 s). Hard syncing clocks.
[ WARN] [1502152360.698515726]: PR: request param #23 timeout, retries left 2, and 670 params still missing
[ WARN] [1502152361.709769004]: PR: request param #24 timeout, retries left 2, and 669 params still missing
[ WARN] [1502152362.755477519]: PR: request param #25 timeout, retries left 2, and 668 params still missing
[ WARN] [1502152363.768851584]: PR: request param #26 timeout, retries left 2, and 667 params still missing
[ WARN] [1502152364.809912862]: PR: request param #27 timeout, retries left 2, and 666 params still missing
[ WARN] [1502152365.858290706]: PR: request param #28 timeout, retries left 2, and 665 params still missing
[ WARN] [1502152366.907638727]: PR: request param #29 timeout, retries left 2, and 664 params still missing
[ INFO] [1502152367.293376845]: HP: requesting home position
[ WARN] [1502152367.949279123]: PR: request param #30 timeout, retries left 2, and 663 params still missing
[ WARN] [1502152368.991500223]: PR: request param #31 timeout, retries left 2, and 662 params still missing
[ WARN] [1502152370.010857838]: PR: request param #32 timeout, retries left 2, and 661 params still missing
[ WARN] [1502152371.052194162]: PR: request param #33 timeout, retries left 2, and 660 params still missing
[ WARN] [1502152371.812381378]: TM: Clock skew detected (0.017690092 s). Hard syncing clocks.
[ WARN] [1502152372.094161292]: PR: request param #34 timeout, retries left 2, and 659 params still missing
[ WARN] [1502152373.110439730]: PR: request param #35 timeout, retries left 2, and 658 params still missing
[ WARN] [1502152374.153971006]: PR: request param #36 timeout, retries left 2, and 657 params still missing
[ WARN] [1502152375.195110337]: PR: request param #37 timeout, retries left 2, and 656 params still missing
[ WARN] [1502152376.210184052]: PR: request param #38 timeout, retries left 2, and 655 params still missing
[ WARN] [1502152377.257989453]: PR: request param #39 timeout, retries left 2, and 654 params still missing
```

If I stop mavlink-router and connect Mavros directly to the serial port these errors go away.

zehortigoza, 2017-10-16:
@lamping7 Did you set this at each boot? `echo 4 > /sys/class/tty/ttyS1/rx_trig_bytes` also try to run mavlink-router with "-r" parameter to check the dropout of messages.

lamping7, 2017-10-16:
@zehortigoza Yes. If you look ^^^^^ you'll see the `ExecStartPre` line. 

I've narrowed it down a bit. I got the errors param request errors to go away by closing QGC. So, it seems like there may be an issue with mavlink-router. 

However, here is the status from mavlink-router you requested, at 921600, which isn't very good.
```
Endpoint UART [7] {
	Received messages {
		CRC error: 333 0% 62KBytes
		Sequence lost: 45050 14%
		Handled: 276083 31344KBytes
		Total: 321466
	}
	Transmitted messages {
		Total: 12975 319KBytes
	}
}
```
That is with QGC closed and Mavros running via TCP connection.

zehortigoza, 2017-10-16:
Here running with the mavlink-router of the last stable release(2de87d42501665df1ef7e707db23e1d25b8df245) + PX4 master(cc724438f930cae064edd67be7dc18500c6bd409) + QGC I got this:
```
Endpoint UART [7] {
        Received messages {
                CRC error: 16 0% 2KBytes
                Sequence lost: 1579 5%
                Handled: 28024 3150KBytes
                Total: 29619
        }
        Transmitted messages {
                Total: 272 4KBytes
        }
}
```

Then overtime it starts to lose less messages in %:
```
Endpoint UART [7] {
        Received messages {
                CRC error: 59 0% 8KBytes
                Sequence lost: 4010 3%
                Handled: 97028 11217KBytes
                Total: 101097
        }
        Transmitted messages {
                Total: 430 9KBytes
        }
}
```

But your setup is different because Mavros send a lot more messages than QGC.

What could be wrong in mavlink-router is the message routing algorithm causing the responses never get back to mavros nodes and that would example why it works when you run mavros directly over serial port.
Can you share your Docker mavros setup? Dockerfile? So I can try to reproduce and check if is a routing problem in mavlink-router.

lucasdemarchi, 2017-10-16:
@lamping7 it's probably due to q bug in mavlink-router that has been fixed. Try mavlink-router from master branch

lucasdemarchi, 2017-10-16:
I'm talking about this fix: https://github.com/01org/mavlink-router/commit/ad367033fd75f1535c82d2e604c7e6be93e16ad5

lamping7, 2017-10-16:
@zehortigoza The mavlink-router I have is whatever comes with OS version 1.5.1 -- what commit is this? I used same the same px4 master. 

I set this up last week, so the version should be alike mavros 0.21.2, ros kinetic, ubuntu 16.04. I'm not in the lab now -- so:
`docker pull ros`
`docker run -it --priveleged --net=host ros`

I don't think it comes with mavros -- so.
`sudo apt install ros-kinetic-mavros`
If you haven't done this recently, you'll find they now depend on a geolib, you you'll have to run the install script. See https://github.com/mavlink/mavros/blob/master/mavros/scripts/install_geographiclib_datasets.sh

Then launch mavros with the docker address and tcp port for the fcu_url.

lamping7, 2017-10-16:
@lucasdemarchi Do you have a guide for updating the mavlink-router on the Intel Areo?

lucasdemarchi, 2017-10-16:
1) Clone the repo in your machine; 2) Follow the readme to build it; 3) copy the mavlink-routerd binary to aero

lamping7, 2017-10-16:
I looked at the readme -- that look fairly straightforward. 

Also looking at the issue you linked to, I can see that causing the param request issue when QGC was open, but I also had 14% loss with only `FC <--> mavlink-router <--> mavros` running, no other apps connected. I don't see how that would explain the latter, but I can give it a shot.

RohanKurane, 2019-01-02:
Is there a solution for this ? I have updated to the latest including Ubuntu 16.04

lucasdemarchi, 2019-01-02:
Not a solution. Just a mitigation that is to run on slower rate and setting the threshold of the uart. This is done by the default configuration of mavlink-router.