intel-aero/meta-intel-aero #25: Update meta-ros to latest version + include librealsense in meta-ros

mindThomas, 2017-03-16:
Hi.
We have been struggling for weeks now with a lot of technical issues with the Intel Aero drone and the "default" Yocto installation that comes with it (and released here on Github).
I am here trying to mark down our findings and issues.
 - Old MAVROS version which does not support Ulogger
 - Old meta-ros version
 - meta-ros layer does not include librealsense

Finally we will now be trying to install Ubuntu as we are simply giving up on these Yocto issues!

Regards Thomas

vilvo, 2017-03-17:
I synced meta-ros fork in aero and change the dev-manifest to point to the latest in it.

Would have tested the build but our AWS build server at http://ec2-52-59-242-235.eu-central-1.compute.amazonaws.com:8080/ ran out of disk space, need to check that and the build the next week.
If you plan to build and test, please not that dev-manifest also points to newer kernel which currently drops some Aero patches.

mindThomas, 2017-03-19:
Can you please elaborate on which kernel patches this include?
I guess we could also just take the default manifest and change the meta-ros revision to the newly update/latest revision?
So there is no need for applying any Aero specific patches when it comes to meta-ros? Is this forked straight from the meta-ros repo without making any changes?

vilvo, 2017-03-20:
Sure, you can see the list of kernel patches in this commit: https://github.com/intel-aero/meta-intel-aero/commit/0cd3ca13f7bba8ef9dd3c960530496a16c5a2b5a

To your proposal - I just tried to build the dev-manifest with the updated meta-ros but it introduces also librealsense-support which causes an error with librealsense recipe from the release manifest:
`ERROR: Multiple versions of librealsense are due to be built (/workspace/poky/meta-intel-aero-librealsense/recipes-support/librealsense/librealsense_1.9.4.bb /workspace/poky/meta-ros/recipes-extended/librealsense/librealsense_1.11.2.bb). Only one version of a given PN should be built in any given build. You likely need to set PREFERRED_VERSION_librealsense to select the correct version or don't depend on multiple versions.`
Any resolution of this requires testing with Aero and may affect different users differently. Need to look into it in more detail.

On meta-ros, with patches I was referring to Aero kernel specific patches, not meta-ros related.

mindThomas, 2017-03-20:
Yes, this was also the exact problem we were facing.
But why is it that we are unable to build any ROS applications then (catkin build) when we try to include this apparently detected librealsense library? (using the current default Intel Aero image)
When we include it as a dependency in our make configuration we get an error stating that it is unable to locate the cmake config for librealsense.

lucasdemarchi, 2017-03-28:
#36 and #38 tasks to be accomplished for M0.

mindThomas, 2017-03-28:
Thanks. Looking forward to that release.

mbelluzzo, 2017-04-04:
I will post a status update here leaving off problems caused by side effects as discussed on (#61). From our last build, librealsense samples work, dmesg complains about not finding R200-in_udev file but works. Since now we will track meta-intel-realsense upstream. I'm investigating why of this (if it is being installed elsewhere, if not or if its no longer needed but not everything update accordingly), open an issue on upstream repo and try to address a temporary fix for intel-aero.

mindThomas, 2017-04-07:
Tested out successfull build by Jenkins (intel-aero-image-intel-aero-20170406055649.iso) today on our Intel Aero drone.
Booted successfully and contained librealsense and the realsense-camera ROS node.
We were able to launch the RealSense ROS nodelet with just RGB and Depth image: 
`roslaunch realsense_camera r200_nodelet_default.launch`
This successfully loads and we are able to see the image using eg. RVIZ.

But we are unable to launch the ROS nodelet resulting in the point cloud:
`roslaunch realsense_camera r200_nodelet_rgbd.launch`
Here we get the following error:
`ResourceNotFound: rgbd_launch`
The full error output is shown below:
```... logging to /home/root/.ros/log/0695d98a-1a9d-11e7-aaf3-8ac716dcb7ee/roslaunch-intel-aero-2118.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/__init__.py", line 307, in main
    p.start()
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 746, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 718, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 652, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 682, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 587, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/substitution_args.py", line 316, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/substitution_args.py", line 329, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/substitution_args.py", line 142, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/substitution_args.py", line 188, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/site-packages/rospkg/rospack.py", line 200, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: rgbd_launch
ROS path [0]=/opt/ros/indigo
ROS path [1]=/opt/ros/indigo/share
```

lucasdemarchi, 2017-04-11:
@mindThomas Thanks for testing. We can leave #82 to track the remaining issue.

closed 2017-04-11 by lucasdemarchi