intel-aero/meta-intel-aero #26: Provide automounting solution for microSD card

lfelipe, 2017-03-16:
This will be needed in case we want to provide a solution that also stores flight logs directly in a microSD card with the user just "plugging it in".

vilvo, 2017-03-17:
It´s a good use case. Safe umount for user to "unplug it" is trickier so we would need to define acceptable criteria. Can we rely on flight logs writing syncing regularly and do you see a risk of corrupting the data if user feels lucky and just "unplugs it"?

lucasdemarchi, 2017-03-28:
We do sync it regularly for other reasons... we could be facing a very imminent crash and we do want the logs in that case. 

@lfelipe systemd provides a very easy way to do it so I'm going to put it under the systemd umbrella.

vidhyagithub, 2017-05-24:
The possible solution would be to define UDEV rule for capturing SD card insert event and to start custom script for copying the logs. Please validate.

diksha-gupta, 2017-05-26:
The following UDEV criteria might do the needful.

1. Create a new rule in /etc/udev/rules.d/ path as below.
ACTION=="add", KERNELS=="mmc2:e624", SUBSYSTEMS=="mmc", ATTRS{type}=="SD", RUN+="/usr/local/bin/script_name.sh"

2. script_name.sh file should have the logic for mounting/un-mounting and redirecting logs into SD card.

sugnanprabhu, 2017-06-16:
@lucasdemarchi can you please review the PR #172 which uses systemd to mount micro sdcard. I tested the changes on aero and able to see that the card gets mounted to /media/sdcard

Changes are based on intel-edison
http://git.yoctoproject.org/cgit/cgit.cgi/meta-intel-edison/tree/meta-intel-edison-distro/recipes-core/base-files

closed 2017-06-23 by lucasdemarchi