intel-aero/meta-intel-aero #34: Runtime configure the RTF FGPA to act as recovery-fpga

lucasdemarchi, 2017-03-28:
Similar to #33, but we could combine both firmwares so people don't have to flash a new one.

lucasdemarchi, 2017-07-18:
@zehortigoza could you add an ISR to PX4 and ArduPilot so we can use a gpio for that task? It looks like even after the pin "reshuffle" to get an additional I2C we could use PC15 from the stm32 for that.  This is something we could put together with the next fpga version bump.

zehortigoza, 2017-07-18:
But how we will control the GPIO state? save the state in FPGA would require a lot of work.

lucasdemarchi, 2017-07-18:
We don't need to save the state. This doesn't require reboot. On the stm32 it's basically:

```c++
static int board_reset_irq(int irq, FAR void *context)
{
    if (stm32_read_gpioread(RESET_GPIO) {
        up_mdelay(RESET_DEBOUNCE);
        if (stm32_read_gpioread(RESET_GPIO) {
            px4_systemreset(false);
        }
    }
}
```

This is only a sketch. The mdelay there may not be ideal/allowed and we should rather have a timer irq firing 5msec later (but we don't want to depend on the rest of the system working): if at least the irq for that specific pin is working, we can reset the microcontroller in runtime.

An alternative in order not to sleep on the irq and not depend on a timer irq is to make the FPGA generate a PWM with e.g. 10Hz, 50% duty cycle. Then the irq would read the gpio and check we stay in that state for 1 or 2 sec.

If aerofc is going to be back alive later without a clean reboot it's something to investigate.

lucasdemarchi, 2017-07-18:
On the FPGA side:

We always initialize the pin with 0 (or 1, should match the irq handler). When we give a spi command  AEROFC_REBOOT from Linux, we toggle this pin (or make the PWM above for X sec).

zehortigoza, 2017-07-18:
Ok but keep in mind that in some cases(like trap in loop inside of a interrupt handler) it will not work and will require power off and power on of the board.

lucasdemarchi, 2017-07-18:
Yep, this is something that should cover most of the cases before telling the user to flash a different fpga firmware and go through the power cycles.

zehortigoza, 2017-07-25:
FPGA: https://github.com/intel-aero/intel-aero-fpga/pull/4
Ardupilot: changes merged
PX4: change submitted https://github.com/PX4/Firmware/pull/7682 
Intel Aero image: missing send the updated jam and add a script to set/unset the force bootloader pin in runtime.

zehortigoza, 2017-07-27:
Done

closed 2017-07-27 by zehortigoza