intel-aero/meta-intel-aero #37: Add ROS node for realsense

lucasdemarchi, 2017-03-28:


lucasdemarchi, 2017-03-29:
This was merged in the dev branch, but it broke the build since the version and patches from the meta-ros layer conflict with the one on meta-intel-aero-librealsense. sigh.

meta-intel-aero-librealsense eventually will go away after the layer cleanups, but maybe fixing it for now is just a matter of understanding which place has correct version of kernel patches and make the version on this meta layer just a .bbappend.

@mbelluzzo can you take a look on it?

mbelluzzo, 2017-03-29:
meta-ros upstream (they dont keep releases) doesn't ship librealsense anymore, actually it doesn't ship some recipes for python tools as well, it relies on meta-intel-librealsense and meta-python (after a given version). I will try to bump everything to match their newest versions/releases and everything should work.

closed 2017-04-01 by lucasdemarchi