intel-aero/meta-intel-aero #41: Enable seemless flight stack logging

lucasdemarchi, 2017-03-28:
Mavlink Router already has capability to save the flight stack log (for PX4 is done, for ArduPilot is wip). Enable it by default on BSP.

lucasdemarchi, 2017-04-01:
Done on mavlink-router. Just needs updated recipe.

lucasdemarchi, 2017-04-01:
ok, #38 can be used for tracking the update.

closed 2017-04-01 by lucasdemarchi

mindThomas, 2017-05-05:
But there is no seamless logging. There is a lot of dropouts if you start digging into the Ulog files or look at the terminal output from the logger.

lucasdemarchi, 2017-05-05:
Dropouts are not related to who is saving the logs, but rather px4 not having enough bandwidth for what it's trying to save.

mindThomas, 2017-05-05:
Alright. So your definition of 'seemless' does not include dropouts?

lucasdemarchi, 2017-05-05:
You are commenting on an old issue/task that was related to adding the support for saving logs on the Linux side integrating the implementation done on mavlink-router.

There are issues with it that are not related with how we are saving the logs and there's an open issue about HSUART1 already.

lucasdemarchi, 2017-05-05:
"Seamless" there was in the sense of how it's integrated... Maybe another word would have been better.