intel-aero/meta-intel-aero #42: Use meta-intel

lucasdemarchi, 2017-03-28:
We should be using meta-intel layer so 1) we share all the improvements from that layer and 2) It makes our lives easier wrt updates.

- meta-intel-aero should be a thin layer, basically for configuration
- meta-intel-aero-* can probably be removed and the resective layers from yocto could be forked directly. This way it's easier to see what's different from baseline and intel-aero
- Bump kernel to meta-intel: 4.4, 4.9 or 4.10

lucasdemarchi, 2017-03-28:
@whbruce any comment on how the layers on meta-intel-aero are structured?  Currently for each "fork" of a layer from yocto there's a meta-intel-aero-<layer> layer added on top with the customizations. This makes it really difficult to keep track of updates and what's coming from where.  Do you agree we should instead fork the corresponding layer and add things on top?

First example that comes to mind is the update of linux-firmware so we get the Core24 for iwlwifi. This is already done in yocto... we should be able to just backport the related patches rather than copying the recipes to a different layer.

lucasdemarchi, 2017-03-28:
M1, but possibly the next one.

whbruce, 2017-03-28:
Definitely a good idea to move to meta-intel BSP. You'll get better compiler tunings for compute intensive workloads and updated iwlwifi microcode (8000C-19). 

Do you want to keep the intel-aero machine name, or move a different one that reflects meta-intel nature of the new machine definition?

Let's keep layer clean-up as a separate issue.

lucasdemarchi, 2017-03-28:
I don't care much about the machine name, although I think leaving as intel-aero would be good.

Hostname definitely needs to be intel-aero as we use it as a link local name.

lucasdemarchi, 2017-03-28:
I updated the description and created a separate issue for layer cleanup:
https://github.com/intel-aero/meta-intel-aero/issues/43

whbruce, 2017-04-08:
@lucasdemarchi, you can assign this to me.

lucasdemarchi, 2017-04-10:
@whbruce I'm thinking that this should be handled together with #43. What do you thinking of creating a morty branch and we make it from scratch? Meaning... the initial setup is just to boot the board using meta-intel, likely using the core-image-minimal.  Then we start to bring in features from the master branch.

If for next release we decide to base off 2.3, we just port to this other branch in a clean way rather than from scratch. What do you think?

whbruce, 2017-04-10:
I have a couple of recommendations 

- I'd like to separate using meta-intel (minor task) from changing Yocto version (major task)
- Once meta-intel is in place I propose we skip morty and go straight for 2.3 (currently master, due for release in May) as meta-intel will then have kernel 4.9LTS so gets SoC driver backports. Morty kernel 4.8 is not LTS.

lucasdemarchi, 2017-04-10:
@whbruce I wouldn't like to postpone the cleanup of layers and the way the distro is done based on "let's way the next yocto release". Releases can slip. If we wait for the release then we will delay a lot the tasks on M1. If done right, jumping from 2.2 to 2.3 should be fairly easy and take a day of work.

The way you are proposing, how long do you think it will get to have meta-intel working?

whbruce, 2017-04-11:
This will take me a day or two. I'll start with kernel 4.4.3 and update to 4.4.26 when you can add the reabsed patches.

closed 2017-04-14 by lucasdemarchi