intel-aero/meta-intel-aero #43: Cleanup layers

lucasdemarchi, 2017-03-28:
meta-intel-aero-* can be removed and the respective layers from yocto could be forked directly. Or depending on the package we could just have some .bbappend files. This way it's easier to see what's different from baseline and intel-aero

lucasdemarchi, 2017-04-19:
First step is to move recipes from meta-intel-aero-connectivity to meta-intel-aero. @avinash-palleti can you take a look on this?

avinash-palleti, 2017-04-19:
@lucasdemarchi, i will look into it

avinash-palleti, 2017-04-19:
@lucasdemarchi, Can you check above patch. If it is Ok, i will send pull request.

I have kept recipes and bbappend files as it is into intel-aero. If required we can merge bbappend into original recipe files as well. for eg. images.bbappend from connectivity can be merged in images.bb of intel-aero.

lucasdemarchi, 2017-04-19:
images.bbappend can probably be removed and merged into images.bb.

I didn't do the diff, but if you copied the files from their current location to meta-intel-aero, it's probably correct.

What still need to be done:

1. remove meta-intel-aero-connectivity from the manifest
1. add a commit to meta-intel-aero-connectivity doing the following:
    * remove everything
    * add a README.md saying content has been moved to meta-intel-aero

whbruce, 2017-04-21:
Ideally hardware specific (i.e. BSP) recipes should be moved to a separate layer.

lucasdemarchi, 2017-05-16:
on master-next

closed 2017-05-16 by lucasdemarchi