intel-aero/meta-intel-aero #44: Review packages installed

lucasdemarchi, 2017-03-28:
Review current list of packages on the image and remove unnecessary cruft. There are lots of packages installed that aren't used anymore and shouldn't be installed. And on the other hand there are useful packages like psutils that aren't.

lucasdemarchi, 2017-04-07:
This may be useful:
http://www.thelins.se/johan/blog/2014/06/yocto-part-ii-baseline-image-size/

avinash-palleti, 2017-04-13:
100802	KiB linux-firmware
73917	KiB opencv-apps
60302	KiB gcc
36539	KiB mesa-megadriver
31581	KiB pymavlink
30161	KiB ros-mavlink
24462	KiB libicudata56
23230	KiB libopencv-imgproc3.1
18786	KiB python-tests
17392	KiB cmake
15689	KiB opencv-staticdev
15478	KiB python-numpy
14837	KiB binutils
11810	KiB kernel-firmware-shisp-2401a0-v21
10585	KiB linux-firmware-iwlwifi-8000c
10491	KiB libopencv-core3.1
10238	KiB gtk+3-dev
9730	KiB nss
8978	KiB libgtk-3.0
8781	KiB linux-firmware-wl12xx
8242	KiB libgcc-s-dev
8174	KiB libstdc++-dev
8033	KiB linux-firmware-iwlwifi-7265d
7485	KiB gobject-introspection-dev
6950	KiB kernel-image-4.4.3-yocto-standard
6939	KiB linux-firmware-radeon
6922	KiB airmap
6517	KiB gdb
6353	KiB python-pip
6286	KiB linux-firmware-iwlwifi-8265
6234	KiB linux-firmware-iwlwifi-7260
6135	KiB libgtk-2.0
6105	KiB linux-firmware-iwlwifi-7265
6067	KiB linux-firmware-ath10k
6058	KiB coreutils
4509	KiB swig
4337	KiB libc6
4218	KiB linux-firmware-adsp-sst
4198	KiB opencv-dev
4145	KiB icu-dev
3929	KiB file
3659	KiB rpm-libs
3588	KiB python-codecs
3587	KiB linux-libc-headers-dev
3554	KiB opencv-samples
3544	KiB gstreamer1.0-dev
3290	KiB bluez5
3253	KiB libopencv-tracking3.1
3204	KiB perl-dev
3066	KiB networkmanager
2934	KiB libglib-2.0-0
2916	KiB modemmanager
2822	KiB mavros
2758	KiB xkeyboard-config
2701	KiB python-misc
2680	KiB gettext
2571	KiB gstreamer1.0-plugins-base-dev
2529	KiB gstreamer1.0
2478	KiB libicui18n56
2444	KiB libgphoto2-6
2433	KiB dhcp-client
2419	KiB libc6-dev
2381	KiB gobject-introspection
2371	KiB protobuf
2325	KiB libcrypto1.0.0
2298	KiB xserver-xorg
2263	KiB libgphoto2-camlibs
2259	KiB shared-mime-info
2185	KiB python-distutils
2153	KiB libopencv-imgcodecs3.1
[... 1761 lines cut ...]
0	KiB initscripts-dev
0	KiB gstreamer1.0-rtsp-server-meta
0	KiB gstreamer1.0-plugins-good-meta
0	KiB gstreamer1.0-plugins-good
0	KiB gstreamer1.0-plugins-base-meta
0	KiB gstreamer1.0-plugins-bad-meta
0	KiB gstreamer1.0-meta-x11-base
0	KiB gstreamer1.0-meta-base
0	KiB glibc-gconvs
0	KiB gcc-symlinks
0	KiB g++-symlinks
0	KiB fbset-modes
0	KiB diffutils-dev
0	KiB cryptodev-linux
0	KiB cpp-symlinks
0	KiB coreutils-dev
0	KiB cantarell-fonts-dev
0	KiB boost
0	KiB bc-dev
0	KiB bash-dev
0	KiB bash-completion-dev
0	KiB base-passwd-dev
0	KiB base-passwd
0	KiB base-files-dev
0	KiB alsa-states

avinash-palleti, 2017-04-13:
@lucasdemarchi Above is the list of packages and size in final image

whbruce, 2017-04-14:
There are some developer tools and libraries in there; gcc, make, cmake, *-dev packages etc. I advise that you create two images, one with developer tools and one without. Build SDKs against images without developer tools otherwise you'll get unnecessarily large installers. SDKs will never use tools "in" the image as an SDK toolchain is built as a separate task and included in the installer. 

The Yocto Project recommends that you use a cross developnent model, but having two images customers can chose which model they want.

lucasdemarchi, 2017-04-14:
@whbruce agreed. I also think linux-firmware got bloated on the last release when we updated the iwlwifi. It's bringing all the unnecessary firmwares with it.

avinash-palleti, 2017-04-18:
@whbruce 

i tried to cherrypick only required firmwares to reduce the size. I created bbappend file for linux-firmware as suggested in below link,

http://yocto.yoctoproject.narkive.com/9VXfWQoF/installing-only-part-of-linux-firmware

But i am getting below error,
ERROR: linux-firmware-1_0.0+gitAUTOINC+c555311c5a-r0 do_package: QA Issue: linux-firmware-iwlwi                 fi-8000c is listed in PACKAGES multiple times, this leads to packaging errors. [packages-list]

closed 2017-05-16 by lucasdemarchi