intel-aero/meta-intel-aero #45: Add collision avoidance library

lucasdemarchi, 2017-03-28:
https://github.com/01org/collision-avoidance-library

mbelluzzo, 2017-03-28:
- [ ] Move 'meta-coav' from coav's repo (we need a public repo for that)
- [ ] Add 'meta-coav' on meta-intel-aero (I have a scratch of this already)

lucasdemarchi, 2017-03-29:
No more meta-*. Add it to meta-intel-aero

mbelluzzo, 2017-03-29:
No way, coav needs its meta layer for its own reason. Also, coav isn't a product of intel aero, it makes no sense to maintain recipes to it there. We may have a custom recipe on meta-intel-aero that makes usage of meta-coav.

lucasdemarchi, 2017-03-29:
If you are going to maintain a meta-coav, fine. I think it would be better if you go through the yocto's  standard layers than to keep your own layer for a single package. This would avoid problems like the one that happened for meta-librealsense vs meta-ros.

lfelipe, 2017-03-29:
Something like meta-drones which includes mavlink-router, coav, camera streaming and possibly others would make sense. Then it would be easy for others to integrate this as a set (and it would serve as a location to integrate newer components as well).

avinash-palleti, 2017-06-01:
@lucasdemarchi Shall i take up this? Is there any plan in place for this?

anselmolsm, 2017-06-05:
@avinash-palleti PR #162

closed 2017-06-16 by anselmolsm