intel-aero/meta-intel-aero #47: Review/improve documentation on the public wiki

lucasdemarchi, 2017-03-28:
There are lots of outdated info on our wiki. And they will become even more after M0 and M1 are completed.  Review and update the wiki after each milestone.

whbruce, 2017-03-28:
The [Developing on Intel Aero](../wiki/Developing-on-Intel-Aero)  section needs particular attention and should reference use of Extensible SDK as per #15

lucasdemarchi, 2017-03-28:
For flashing the thumb drive, besides the dd info we should add Etcher, that's a graphical utility available for Linux, Windows and Mac.

We can think about bmaptool, too which is much faster than any other method, but I think it's Linux-only.

lfelipe, 2017-04-05:
@guermonprez just updated the wiki with a new version. It needs to be updated to take under consideration the modifications done for Milestone M0. 

- [ ] New update procedure;
- [ ] Information about logs;

whbruce, 2017-04-05:
[Developing-on-Intel-Aero](../wiki/Developing-on-Intel-Aero) needs attention. We should not be encouraging app development in the bitbake environment. Instead we should promote use of the extensible SDK and devtool. I would be happy to contribute the changes.

lucasdemarchi, 2017-04-14:
I just updated several wiki pages to give instructions on  the  new update procedure, disambiguate and correct some phrases. The page about using pymavlink/dronekit I think needs to be redone, so I deleted it.

@guermonprez we use mavlink-router instead of mavproxy. C and python examples on how to communicate with the flight stack are available on that repo. An example with dronekit there would be greatly appreciated.

closed 2017-04-14 by lucasdemarchi