intel-aero/meta-intel-aero #50: Tools that should not be provided by busybox

lucasdemarchi, 2017-03-30:
This is a collection of tools that we keep adding here that should be provided by the original tools, not the cap'ed ones from busybox.
- ps
- tar

Just because it's horrible to use them and this should be developer-friendly.

avinash-palleti, 2017-04-04:
[packages.xlsx](https://github.com/intel-aero/meta-intel-aero/files/893320/packages.xlsx)

Lucas,
Please find the data on packages which are from busybox.

avinash-palleti, 2017-04-05:
@whbruce 
One query, as per below link the one with highest priority will go 
into final image.
 http://stackoverflow.com/questions/34609716/yocto-busybox-overriding-coreutils
But in case of tar & ps, i see their priority in meta/recipes-extended/tar and meta/recipes-extended/procps as 100 and 200 respectively, where as meta/recipes-core/busybox priority is 50 then why in final image these utilities came from busybox though its priority is less.

PS: meta layer is included in BBLAYERS.

whbruce, 2017-04-08:
Your approach is not working as `tar` and `ps` are not provided by `coreutils` but their own recipes. You can determine this by looking at the content of `ALTERNATIVES` variable in the `coreutils` recipe and seeing they are not present.

Instead simply add the packages to the [core-tools packagegroup](../recipes-core/packagegroups/packagegroup-core-tools.bb). This works as both the `tar` and `procps` recipes set `ALTERNATIVE_PRIORITY` higher than busbox.

```
RDEPENDS_${PN} = "\
		..
		tar \
		procps \
		"
```

avinash-palleti, 2017-04-10:
Thanks @whbruce for response. first approach will not work as these utilities are not provided by coreutils.
http://www.catonmat.net/download/gnu-coreutils-cheat-sheet.pdf

I tried second approach and it is working. Added tar and procps in core-tools packagegroup.

avinash-palleti, 2017-04-10:
@lucasdemarchi I have submitted pull request to update tar and ps utilities. Now from the excel sheet i attached above, we need to decide which tools has to be updated.

lucasdemarchi, 2017-04-10:
@avinash-palleti thanks.

Put the table here in plain text so it's easier to comment on them rather than using a spreadsheet

avinash-palleti, 2017-04-10:
Utilities from busybox:

addgroup
adduser
ash
blkid
clear
cpio
delgroup
deluser
dnsdomainname
dumpkmap
dumpleases
free
fsck
fuser
getopt
getty
gunzip
gzip
hostname
hwclock
ifconfig
ifdown
ifup
ip
killall
klogd
less
loadfont
loadkmap
logread
microcom
nc
netstat
nslookup
patch
ping
ping6
ps
rdate
reset
resize
route
run-parts
setconsole
start-stop-daemon
sysctl
syslogd
tar
telnet
tftp
time
top
traceroute
udhcpc
udhcpd
unzip
usleep
vi
vlock
watch
wget
which
zcat

lucasdemarchi, 2017-04-11:
@avinash-palleti thanks for the list. I think the one to move right now is `top`. The others can remain as is for now.

avinash-palleti, 2017-04-11:
@lucasdemarchi "top" utility is also from procps-ng
Above patch should include "top" utility also, will confirm tomorrow after checking on final image.

avinash-palleti, 2017-04-12:
@lucasdemarchi 
Following utilities are already from procps-ng after merging above patch,
free
ps
sysctl
top
watch

closed 2017-04-12 by lucasdemarchi