intel-aero/meta-intel-aero #52: change default startup from ap_mode to station mode

blankered, 2017-03-31:
Hi,
 I am trying to determine the most minimally invasive way of switching the default ap_mode on boot to station mode. I have found some sources on system_ctl but I am unfamiliar with that tool. Any help is greatly appreciated.

lucasdemarchi, 2017-03-31:
Have you seen https://github.com/intel-aero/meta-intel-aero/wiki/Functionality#station-mode ? It may change slightly for the next release, but it's currently what's there.

blankered, 2017-04-03:
Yes I have reviewed this documentation. I am looking for documentation that will allow me to boot into station_mode rather than boot into AP_Mode. This would reduce the needed process of logging into the Aero while it is in AP_Mode, running the autostart-supplicant.sh script, waiting for the Aero to switch over, logging back in on my own network. I think I need to append an rc script or another boot script?

closed 2017-04-05 by blankered