intel-aero/meta-intel-aero #61: Contradictory manifest and BBLayers.sample file

mindThomas, 2017-04-02:
After recent commits (eg. 3a09bb0c0280a024a5b50683b1467ba962e63866) the default manifest file and the BBLayers file are no longer matching.

The result is a broken build.
OBS. All new users trying to build will not be able to follow the guide and build.

lucasdemarchi, 2017-04-02:
It should be meta-intel-realsense. I think I overlooked it while merging stuff on Friday. Sorry.

mindThomas, 2017-04-03:
Not quite. Just renaming gives you the following error:
```
ERROR: No recipes available for:
  /home/thomas/intel_aero/poky/meta-intel-aero/recipes-support/librealsense/librealsense_1.12.1.bbappend
```
This might be explained by the fact that the manifest points to the meta librealsense layer hosted within the intel-aero git, which has not yet been updated to version 1.12.1 (see https://github.com/intel-aero/meta-intel-aero-librealsense/tree/krogoth/recipes-support/librealsense)

mbelluzzo, 2017-04-03:
There was a change on the manifest pointing to the new repo. You will have sync/init from the newest manifest. Dont forget to export TEMPLATECONF before sourcing poky env.

mindThomas, 2017-04-03:
I have tried this on a completely clean/fresh installed machine today, taking the latest manifest using the Getting Started guide (as I've done always).
This does not work!

mindThomas, 2017-04-03:
And just to prove you wrong, take a look at your own Jenkins server output from its' recent build: http://ec2-52-59-242-235.eu-central-1.compute.amazonaws.com:8080/job/aero-build-release/231/console

lucasdemarchi, 2017-04-03:
Humn... The merges were done on Friday and the Getting Started was not updated. So maybe this is the problem... ? We are no longer using the dev manifest for example. Can you post the result of `repo info`?

lucasdemarchi, 2017-04-03:
As per comment in https://github.com/intel-aero/meta-intel-aero/issues/63#issuecomment-291212608, I'll keep this open to track the issue.

mbelluzzo, 2017-04-03:
As I said, there is a dependence between meta-intel-aero and the manifest on our master branches. We had to wait PR https://github.com/intel-aero/meta-intel-aero/pull/56 to be merged (which breaks the build if you use master branch without the changes on the manifest which aren't merged yet) so we can have the manifest PR https://github.com/intel-aero/intel-aero-manifest/pull/5 updated with the revision number after #56 was merged to 
have it merged as well.

Once PR<manifest>#5 gets merged, the build will work again. So if you willing to work on master branch of thoses repositories, you will have to check for unmerged PRs on both places as well because it may be changes that are inter-dependent.

lucasdemarchi, 2017-04-03:
Now I see why I couldn't reproduce it on Friday. The issue is exactly that even though I changed the manifest to point to the correct commits I didn't push it. Doing it now.

mindThomas, 2017-04-03:
Let me just get this right.
When you say "master" branches you are referring to the master branch on the original (non-forked) branch? Or are you referring to just the master branch within the intel-aero repository?

If it is the first case everything makes sense ;)
Yup, I agree that if one make changes and pull requests and another only merges half of the pull requests, you will be asking for trouble.

lucasdemarchi, 2017-04-03:
@mindThomas yep... I merged the changes on all repos, but the changes on the manifest were not pushed for some reason. I just did it now. Can you try to compile it again and see if it's fixed?

mindThomas, 2017-04-03:
I can confirm a successfull build again. I will test the image tomorrow and let you know the outcome.
FIY. Your Jenkins build server is still struggling though: http://ec2-52-59-242-235.eu-central-1.compute.amazonaws.com:8080/job/aero-build-release/lastFailedBuild/console

lucasdemarchi, 2017-04-03:
@mindThomas We just got access to fix the BSP and do this release. We still didn't touch the AWS infra. We will do that soon.

mindThomas, 2017-04-03:
Fantastic. I am very happy that you have managed to merge the changes and get this Intel Aero repo "up-to-date".
I will test and get back with my findings and/or comments.

JochiPochi, 2017-04-03:
I'll jump in to help test this too.
Seems that my PRs won't be happy until this is resolved

JochiPochi, 2017-04-03:
All the recipes were parsed just fine. So far the build is progressing steadily. This issue does seem resolved. I'll report back once I finish testing the image.

mindThomas, 2017-04-04:
Unfortunately no success with the image this morning. Build went fine but it does not boot.
One of the errors noticed along the way of booting the kernel was:
`Failed to execute /usr/local/bin/usb-R200-in_udev`
This was the same error we also had on the build I tried to make last week, when I pulled the latest branches manually.
After this error message the kernel messages continoues and shortly after we get a continous stream of error messages which eventually dies out, ends in a login prompt and soon after (within 5-10 seconds without logging in) the board crashes and restarts.
See an image of the continous stream of error messages here: https://www.dropbox.com/s/nxryj8r8a5d2q4v/2017-04-04%2008.55.13.jpg

lucasdemarchi, 2017-04-04:
@mindThomas I guess you selected the "boot" option rather than leave the default "install".  It seems there's a daemon that's not playing nice with / being read-only or something like when booting the live image.

If you select "install", it should work... but be warned it will erase everything you have on /

lucasdemarchi, 2017-04-04:
https://github.com/intel-aero/meta-intel-aero/pull/57#issuecomment-290884711

These are the new steps for updating without a monitor and keyboard. We are still trying to simplify it a little bit before the release and use the LEDs for user feedback.

lucasdemarchi, 2017-04-04:
I'm closing this issue since we deviated the discussion from the original subject. Feel free to open another one specific to the problem you are having.

closed 2017-04-04 by lucasdemarchi

JochiPochi, 2017-04-05:
I second on starting a separate issue. 
Since I did say I was going to report back after testing the image, for the record:
- I was able to boot without crashing using both "boot" and "install" options on the usb stick's grub menu.
I did see some odd stuff on the terminal, an overwhelming amount of `[[26^` that polluted all output so much that I couldn't see any other errors. But once the GUI came up on the "installed" version, I could use the Aero board as normal.

I will elaborate more on the separate issue if it gets created.

mindThomas, 2017-04-05:
Can I know how and to which type of USB Stick you installed the image?
Did you format the USB stick in any special way before or did you just use the 'dd' command as specified in the Getting Started guide?
We will try the "install" option now and let you know of the outcome.

JochiPochi, 2017-04-05:
Here's a description of my setup:
- Compile machine was an Ubuntu 14.04 x86-64 system
- I use multiple USB sticks for this. A lot of times some cheap free flair I get from conferences and expos.
  When I but my own, I generally use [microcenter USB sticks](http://www.microcenter.com/category/4294966790/usb-flash-drives) but I really don't think that the type of USB stick matters much as long as the size is correct. All of mine are 8GB or larger.
- I used `sudo dd if=tmp/deploy/images/intel-aero/intel-aero-image-intel-aero.iso of=/dev/sdX bs=1M`
  just as the quickstart suggests. Usually I tend to execute `sudo sync` or just `sync` out of habit to make sure data has actually finished writing to flash before taking out the USB stick. I found out that if I pull out USB sticks after using DD sometimes results on corrupt data. Waiting for a few minutes after DD is done would probably accomplish the same thing.