intel-aero/meta-intel-aero #63: Failing: RealSense_Camera_Support_linux-yocto_4.4.patch

mindThomas, 2017-04-03:
With the latest commit 07c3b7ec10daa66c3be7ec8c1ed7a414e5740b65 the RealSense camera patch for the Kernel can not be applied and is failing when building.
Notice I am using librealsense 1.12.1 as supposed with the patch used (see https://github.com/intel-aero/meta-intel-aero/blob/master/recipes-kernel/linux/linux-yocto/realsense_camera_formats_linux-yocto_4.4.patch).
See error log below:
```
|    mark --> gen_homethomasintel_aeropokymeta-intel-realsenserecipes-kernellinuxfiles_desc.scc
|          patch RealSense_Camera_Support_linux-yocto_4.4.patch (.kernel-meta/patches/standard/base/series)
|    mark <-- gen_homethomasintel_aeropokymeta-intel-realsenserecipes-kernellinuxfiles_desc.scc
| (1/1) RealSense_Camera_Support_linux-yocto_4.4.patch
| [INFO]: check of .kernel-meta/patches/standard/base/links/files/RealSense_Camera_Support_linux-yocto_4.4.patch with "git am" did not pass, trying reduced context.
| [INFO]: Context reduced git-am of .kernel-meta/patches/standard/base/links/files/RealSense_Camera_Support_linux-yocto_4.4.patch with "git am" did not work, trying "apply".
| /home/thomas/intel_aero/poky/build/tmp/work-shared/intel-aero/kernel-source/.git/rebase-apply/patch:15: space before tab in indent.
|  	{
| Context reduced to (1/1) to apply fragment at 105
| error: patch failed: drivers/media/usb/uvc/uvcvideo.h:91
| error: drivers/media/usb/uvc/uvcvideo.h: patch does not apply
| Context reduced to (1/1) to apply fragment at 608
| [ERROR]: Application of .kernel-meta/patches/standard/base/links/files/RealSense_Camera_Support_linux-yocto_4.4.patch failed.
|          Patch needs to be refreshed. Sample resolution script:
|              .git/rebase-apply/resolve_rejects
| ERROR. could not update git tree
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_patch (log file is located at /home/thomas/intel_aero/poky/build/tmp/work/intel_aero-poky-linux/linux-yocto/4.4.3+gitAUTOINC+6ec93aaa70_3d2455f9da-r0/temp/log.do_patch.10359)
ERROR: Task 50 (/home/thomas/intel_aero/poky/meta/recipes-kernel/linux/linux-yocto_4.4.bb, do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1500 tasks of which 1413 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:

Summary: 1 task failed:
  /home/thomas/intel_aero/poky/meta/recipes-kernel/linux/linux-yocto_4.4.bb, do_patch
Summary: There was 1 WARNING message shown.
```

mbelluzzo, 2017-04-03:
If you are using master, you should be using master manifest as well, otherwise you will end up with an incompatible build environment. The patch stated by the log is no longer being used on master image. When dealing with the master branch for image infrastructure repos, consider to always upgrade and synchronize everything with 'repo' and our manifest. Also, be aware that the master branch may be broken sometimes due to circular dependencies between the manifest and its layers.

So, unless you are willing to workaround some issues, we recommend you to always build from our releases that are guarantee to be working properly.

closed 2017-04-03 by mbelluzzo

mindThomas, 2017-04-03:
You are clearly closing these issues too quickly.
I am indeed willing to workaround the issues, and with a decent knowledge of Linux, Yocto, ROS etc. I expected that I would come fairly easy around the problems.

I am indeed using the master manifest, pulled today on a fresh install of Ubuntu. Something is not right in your manifest!

lucasdemarchi, 2017-04-03:
@mindThomas that patch should not be applying anymore since it's replaced by another one. I see you opened 3 issues that are probably related. Let's keep #61 open to track down the issue until we figure out what exactly is the problem.  I just started a build from scratch to try to reproduce your issue.