intel-aero/meta-intel-aero #64: Backport new atomisp from upstream kernel

lucasdemarchi, 2017-04-03:
Atomisp has been merged as a staging driver on upstream. Backport to 4.4.x to replace our current version

lkm1321, 2017-07-04:
Hi, I'm trying this out myself just for fun. I have a couple of questions: 
- The current OV8858 driver seems to use struct v4l2_subdev_video_ops to pass the getter/setter functions for format while the mainline driver seems to use struct v4l2_subdev_pad_ops uses v4l2_subdev_pad_ops to do the same. Is this intended?

zehortigoza, 2017-07-05:
@lkm1321 the drivers in Intel Aero kernel is previous to the internal api change in v4l2, I did it to ov7251 https://github.com/zehortigoza/linux-next/commit/6d8d326b1fcd6a10df12403ca7dfebd97ed1ef26