intel-aero/meta-intel-aero #65: How to boot and run intel-aero-image in qemu? Please add instructions to wiki

rachelruijiayang, 2017-04-03:
I have successfully built the Intel Aero image using the instructions in [Quickstart Guide](https://github.com/intel-aero/meta-intel-aero/wiki/Quickstart-Guide#get-sources), and the images are located in ```intel_aero/poky/build/tmp/deploy/images/intel-aero```.  The files in here are:
```
bzImage
bzImage--4.4.3+git0+6ec93aaa70_3d2455f9da-r0-intel-aero-20170403025532.bin
bzImage-intel-aero.bin
core-image-minimal-initramfs-intel-aero-20170403025532.rootfs.cpio.gz
core-image-minimal-initramfs-intel-aero-20170403025532.rootfs.manifest
core-image-minimal-initramfs-intel-aero.cpio.gz
core-image-minimal-initramfs-intel-aero.manifest
intel-aero-image-intel-aero-20170403025532.hddimg
intel-aero-image-intel-aero-20170403025532.iso
intel-aero-image-intel-aero-20170403025532.rootfs.ext4
intel-aero-image-intel-aero-20170403025532.rootfs.manifest
intel-aero-image-intel-aero.ext4
intel-aero-image-intel-aero.hddimg
intel-aero-image-intel-aero.iso
intel-aero-image-intel-aero.manifest
modules--4.4.3+git0+6ec93aaa70_3d2455f9da-r0-intel-aero-20170403025532.tgz
modules-intel-aero.tgz
README_-_DO_NOT_DELETE_FILES_IN_THIS_DIRECTORY.txt
```

**However, is there a way to run the image in the qemu emulator? (The configuration in [meta-intel-aero/conf/local.conf.sample](https://github.com/intel-aero/meta-intel-aero/blob/879de50d861e8e791bf056f4df1d5c8248b4b471/conf/local.conf.sample) seems to provide some settings.)**

I ran ```runqemu qemux86 intel-aero-image``` and got:
```
Assuming intel-aero-image really means /media/ruijia/143d637d-e685-4a48-8f62-c35a040695da/intel_aero/poky/build/tmp/deploy/images/qemux86/intel-aero-image-qemux86.ext4

Continuing with the following parameters:
KERNEL: [/media/ruijia/143d637d-e685-4a48-8f62-c35a040695da/intel_aero/poky/build/tmp/deploy/images/qemux86/bzImage-qemux86.bin]
ROOTFS: []
FSTYPE: [ext4]
Setting up tap interface under sudo
Error: Unable to find tunctl binary in '/media/ruijia/143d637d-e685-4a48-8f62-c35a040695da/intel_aero/poky/build/tmp/sysroots/x86_64-linux/usr/bin', please bitbake qemu-helper-native
ruijia@bliss:/media/ruijia/143d637d-e685-4a48-8f62-c35a040695da/intel_aero/poky/build/tmp/deploy/images/intel-aero$ runqemu qemux86 intel-aero-image
Assuming intel-aero-image really means /media/ruijia/143d637d-e685-4a48-8f62-c35a040695da/intel_aero/poky/build/tmp/deploy/images/qemux86/intel-aero-image-qemux86.ext4

Continuing with the following parameters:
KERNEL: [/media/ruijia/143d637d-e685-4a48-8f62-c35a040695da/intel_aero/poky/build/tmp/deploy/images/qemux86/bzImage-qemux86.bin]
ROOTFS: []
FSTYPE: [ext4]
Setting up tap interface under sudo
[sudo] password for ruijia: 
Error: Unable to find tunctl binary in '/media/ruijia/143d637d-e685-4a48-8f62-c35a040695da/intel_aero/poky/build/tmp/sysroots/x86_64-linux/usr/bin', please bitbake qemu-helper-native
```

lucasdemarchi, 2017-04-04:
Currently there's no support for running it on qemu. This can be added in future releases. Would be a nice addition if anyone is willing to contribute to the project.

lucasdemarchi, 2017-04-11:
@rachelruijiayang currently there's no support for that. After milestone M1 in which this will be using meta-intel layer, there will probably be something to support that.

closed 2017-04-11 by lucasdemarchi