intel-aero/meta-intel-aero #66: how to install nano editor

blankered, 2017-04-03:
While it isn't crucial that I use nano, I wanted to install it as apart of the process of learning to use Yocto. I thought it might be good to post this problem here so others who are looking to develop with Yocto might learn as well. 

Following forum advice for the Intel Edision, I attempted to install nano using the following command:

wget http://www.nano-editor.org/dist/v2.2/nano-2.2.6.tar.gz && tar xvf nano-2.2.6.tar.gz && cd nano-2.2.6 && ./configure && make && make install 

which spits out:

Connecting to www.nano-editor.org (213.138.109.86:80)
Connecting to www.nano-editor.org (213.138.109.86:443)
wget: bad header line: protocol mismatch.

How do I fix this issue? I tried using opkg and smart but they don't have an package sources....

lucasdemarchi, 2017-04-04:
> wget http://www.nano-editor.org/dist/v2.2/nano-2.2.6.tar.gz && tar xvf nano-2.2.6.tar.gz && cd nano-2.2.6 && ./configure && make && make install
>
> which spits out:
>
> Connecting to www.nano-editor.org [...]

You are telling wget to download nano, but your board is not connected to the internet.

If you are building your own BSP, you should rather add it to the packages to be installed in the yocto recipes. See yocto manual for that.

blankered, 2017-04-05:
Thanks for your help!

closed 2017-04-05 by blankered