intel-aero/meta-intel-aero #68: Adopt os-release

lucasdemarchi, 2017-04-03:
https://www.freedesktop.org/software/systemd/man/os-release.html

Better than the house-made os_version

whbruce, 2017-04-12:
Issue #97 means we can set our own distro version (i.e. not inherit poky's version). However it will be useful to retain poky name and version. Perhaps we can use the keys VERSION_CODENAME and VERSION_ID for this purpose.

lucasdemarchi, 2017-04-13:
where would we put our own version? Like... now is 1.2, next is 1.3, 1.4 and so on. We need to be able to reference this number.

whbruce, 2017-04-13:
Yes, you can still have your own version. Config file would look something like this. Look at PR #100 for variable values, they are in `conf/distro/poky.aero.conf`. 

NAME="${DISTRO}"
VERSION="${DISTRO_VERSION}"
VERSION_CODENAME="${DISTRO_CODENAME}"
VERSION_ID="${POKY_VERSION}"
PRETTY_NAME="${DISTRO_NAME}"

closed 2017-04-17 by lucasdemarchi