intel-aero/meta-intel-aero #76: ROS realsense-camera node - undefined symbol: rs_get_detached_frame_data

mindThomas, 2017-04-05:
With the build based on master-branch from 3rd of April I'm not able to launch the realsense-camera node within ROS.
We get the error message shown below, of which it is important to notice this:
> Error string: Could not load library (Poco exception = /opt/ros/indigo/lib//librealsense_camera_nodelet.so: undefined symbol: rs_get_detached_frame_data)

```
roslaunch realsense_camera r200_nodelet_default.launch
... logging to /home/root/.ros/log/5dce8498-18b0-11e7-9238-f2c9c874ab3f/roslaunch-intel-aero-1172.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:33921/

SUMMARY
========

PARAMETERS
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: R200
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/fisheye_frame_id: camera_fisheye_frame
 * /camera/driver/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/driver/imu_frame_id: camera_imu_frame
 * /camera/driver/imu_optical_frame_id: camera_imu_optica...
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir2_optical_frame_id: camera_ir2_optica...
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/ir_optical_frame_id: camera_ir_optical...
 * /camera/driver/mode: manual
 * /camera/driver/serial_no:
 * /camera/driver/usb_port_id:
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /camera/
    driver (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311/

core service [/rosout] found
process[camera/nodelet_manager-1]: started with pid [1190]
process[camera/driver-2]: started with pid [1191]
[ INFO] [1491253182.606289948]: Initializing nodelet with 4 worker threads.
[ERROR] [1491253182.796416681]: Failed to load nodelet [/camera/driver] of type [realsense_camera/R200Nodelet] even after refreshing the cache: Failed to load library /opt/ros/indigo/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/indigo/lib//librealsense_camera_nodelet.so: undefined symbol: rs_get_detached_frame_data)
[ERROR] [1491253182.796510721]: The error before refreshing the cache was: Failed to load library /opt/ros/indigo/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/indigo/lib//librealsense_camera_nodelet.so: undefined symbol: rs_get_detached_frame_data)
[FATAL] [1491253182.797217844]: Failed to load nodelet '/camera/driver` of type `realsense_camera/R200Nodelet` to manager `nodelet_manager'
[camera/driver-2] process has died [pid 1191, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/R200Nodelet nodelet_manager depth:=depth color:=color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/root/.ros/log/5dce8498-18b0-11e7-9238-f2c9c874ab3f/camera-driver-2.log].
log file: /home/root/.ros/log/5dce8498-18b0-11e7-9238-f2c9c874ab3f/camera-driver-2*.log
```

mbelluzzo, 2017-04-05:
Currently we are working with intel-ros team to address that, but they
didn't give us any feedback yet. I will post a status update once I have
news.

icoderaven, 2017-04-05:
Ditto. I think this is super close, and all it requires is just explicit linking in the CMakeLists to the realsense driver for the nodelet. This is the output for debug_info.
 
----------CUT-PASTE-THIS----------
|Version | Your Configuration | 
|:----------|:----------| 
|Operating System | Unable to detect operating system|
|Kernel | 4.4.3-yocto-standard|
|ROS | indigo| 
|ROS RealSense | 1.7.2|
|librealsense | Cannot locate [librealsense]|
|Camera Type | Intel RealSense R200|
|Camera Firmware | 1.0.71.06|
----------CUT-PASTE-THIS----------

mbelluzzo, 2017-04-06:
https://github.com/intel-aero/meta-ros/pull/2

This solved the issue. Now no errors related to librealsense are issued, but there is python error about ros trying to detected the underlining OS. I dont know if this has any impact on the use of this nodelet. Can someone test this as well?

icoderaven, 2017-04-06:
This worked for me!

lucasdemarchi, 2017-04-06:
Change merged on meta-ros and manifest was just pushed.

closed 2017-04-06 by lucasdemarchi