intel-aero/meta-intel-aero #81: "python-imaging-1.1.7-r5 do_package_qa" build problem

mindThomas, 2017-04-06:
Fresh pull of repo yields Python-imaging build errors. Any clue?

```
ERROR: python-imaging-1.1.7-r5 do_package_qa: QA Issue: python-imaging: The compile log indicates that host include and/or library paths were used.
         Please check the log '/home/thomas/intel_aero/poky/build/tmp/work/core2-64-poky-linux/python-imaging/1.1.7-r5/temp/log.do_compile' for more information. [compile-host-path]
ERROR: python-imaging-1.1.7-r5 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: python-imaging-1.1.7-r5 do_package_qa: Function failed: do_package_qa
ERROR: Logfile of failure stored in: /home/thomas/intel_aero/poky/build/tmp/work/core2-64-poky-linux/python-imaging/1.1.7-r5/temp/log.do_package_qa.4200
ERROR: Task 6475 (/home/thomas/intel_aero/poky/meta/recipes-devtools/python/python-imaging_1.1.7.bb, do_package_qa) failed with exit code '1'
```

icoderaven, 2017-04-07:
I mentioned this on my forum post. I manually added the patch at 
https://github.com/bmwcarit/meta-ros/issues/395

lucasdemarchi, 2017-04-07:
I've never seen this build error. And Jenkins as well as our internal build servers are running Ubuntu 16.04.

icoderaven, 2017-04-08:
I got this on 14.04 on a clean build.

mindThomas, 2017-04-08:
Anything special/new that needs to be installed which has not been updated accordingly in the Getting Started guide?
We have now tried in both Ubuntu 14.04 and 16.04 with similar outcome

mindThomas, 2017-04-10:
@icoderaven But "python-imaging_1.1.7.bb", which uses this patch, has been dropped in the latest meta-ros repo (see https://github.com/bmwcarit/meta-ros/issues/395#issuecomment-284385581)
I am currently trying to use the full python recipe folder from your reposity as a replacement (https://github.com/Minipada/meta-ros/tree/9e1a5a452d7d75fce81850a4fc3913751e0d712d/recipes-devtools/python)

mindThomas, 2017-04-26:
OBS. Trying to build today with the master manifest this problem appears again.
The problem was solved by putting in the python-imaging recipe into meta-ros and by applying the patches as found here: https://github.com/Minipada/meta-ros/tree/9e1a5a452d7d75fce81850a4fc3913751e0d712d/recipes-devtools/python/python-imaging

icoderaven, 2017-04-26:
Yeah, likewise.

jaronski, 2018-10-18:
Hi there, i am new to yocto and i am having lots of trouble building meta-ros. I get the same ERROR mentioned above. I don't understand how you solved this error? can someone support me an explain what the steps are to solve this problem?
Thanks

jaronski, 2018-10-19:
@icoderaven thanks for your reply. Unfortunately i have to use Yocto and meta-ros for an industry stereocam system. The System is based on a jetson tx2. The yocto linux is working fine without meta-ros as soon as i put the packagroup-ros-comm into the build i get the error. i changed the meta-ros to the one Minipada supplied without any success. i tried to add the patch manually without any success. Did someone actually solved the error?