intel-aero/meta-intel-aero #82: Realsense-camera ROS Nodelet not launching

mindThomas, 2017-04-07:
Today we tested out the recent successfull build by Jenkins (intel-aero-image-intel-aero-20170406055649.iso) today on our Intel Aero drone. The image boots successfully and contained librealsense and the realsense-camera ROS node.

We were able to launch the RealSense ROS nodelet with just RGB and Depth image: 
`roslaunch realsense_camera r200_nodelet_default.launch`
This successfully loads and we are able to see the image using eg. RVIZ.

But we are unable to launch the ROS nodelet resulting in the point cloud:
`roslaunch realsense_camera r200_nodelet_rgbd.launch`
Here we get the following error:
`ResourceNotFound: rgbd_launch`
The full error output is shown below:
```... logging to /home/root/.ros/log/0695d98a-1a9d-11e7-aaf3-8ac716dcb7ee/roslaunch-intel-aero-2118.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/__init__.py", line 307, in main
    p.start()
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 746, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 718, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 652, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 682, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 587, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/substitution_args.py", line 316, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/substitution_args.py", line 329, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/substitution_args.py", line 142, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roslaunch/substitution_args.py", line 188, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/site-packages/rospkg/rospack.py", line 200, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: rgbd_launch
ROS path [0]=/opt/ros/indigo
ROS path [1]=/opt/ros/indigo/share
```

icoderaven, 2017-04-07:
Just the pointcloud nodelet by itself works, however.

mindThomas, 2017-04-07:
@icoderaven - Can you please elaborate? What pointcloud nodelet are you referring to?

blankered, 2017-04-10:
I had the same problem. You need to compile the rgbd-launch program under the meta-ros layer on your host machine. Then add IMAGE_INSTALL_append = "rgbd-launch" to your conf/machine/intel-aero.conf file under the meta-intel-aero folder. Then rerun bitbake intel-aero-image. Setup your host as a repository using smart if you are building RPM. Install rgbd-launch from your host repo.

mindThomas, 2017-04-10:
@blankered Thanks. I am now able to get the pointcloud working, but only for a few seconds. I am afraid if this error has something to do with what is also mentioned here: https://github.com/intel-ros/realsense/issues/195

mbelluzzo, 2017-04-10:
> Then add IMAGE_INSTALL_append = "rgbd-launch"

I think that the more correct is to add rgbd-launch to RDEPENDS on realsense_camera. We are not using the entire ros package group. That is why I think it was out of our image.

Probably ppl using/development 'meta-ros' are always using 'ros'package group so this went unnoticed by them.

Since we already have a custom patch and a custom recipe waiting for upstream to be bump, I can go ahead and try to bump the version of realsense node itself since 1.8.0 was just considered stable now (last week stable was still 1.7.2), lets see if works

mbelluzzo, 2017-04-10:
@mindThomas @blankered @icoderaven mind to test if this solves the problem for you:

https://github.com/intel-aero/meta-ros/pull/3

closed 2017-04-11 by lucasdemarchi

mindThomas, 2017-04-11:
@mbelluzzo Will be testing tomorrow. At least with 1.7.2 it stops working after a few seconds.

mindThomas, 2017-04-18:
Confirmed working with the latest build now including v1.8.0
But the framerate of the registered pointcloud seem to be dropping right after the node is started. After a while it does not seem to publish anything.
Do we have a memory problem?

icoderaven, 2017-04-19:
Works for me too. Stable 60Hz at default settings. Now if only I could get MAVLink to work...

lucasdemarchi, 2017-04-19:
@icoderaven can you open an issue with details on what's not working?

mindThomas, 2017-04-20:
@lucasdemarchi Did you see my problems with the framerate?