intel-aero/meta-intel-aero #84: Can't connect to MAVlink router from remote computer using dronekit

majassow, 2017-04-07:
I'm having an issue connecting to an Aero platform using Dronekit over wifi (QGroundControl is able to connect to the drone over wifi).

I'm using the command:
vehicle3 = connect(str(args.conn3), source_system=3, wait_ready=False)

if I run this on the Aero platform itself, using args.conn3 = "127.0.0.1:14550" connects successfully.
Using "192.168.1.255:14550" times out:
>>> Link timeout, no heartbeat in last 5 seconds
Using "192.168.1.1:14550" works, but only if I add -e "192.168.1.1" into /etc/init.d/mavlink-routerd and restart.

If I run this command on a different platform connected to wifi on the Aero platform, I cannot connect:
Connecting
Traceback (most recent call last):
  File "./full_mission4.py", line 65, in <module>
    vehicle3 = connect(str(args.conn3), source_system=3, wait_ready=False)
  File "/usr/lib/python2.7/site-packages/dronekit/__init__.py", line 2817, in connect
    handler = MAVConnection(ip, baud=baud, source_system=source_system, use_native=use_native)
  File "/usr/lib/python2.7/site-packages/dronekit/mavlink.py", line 128, in __init__
    self.master = mavutil.mavlink_connection(ip, baud=baud, source_system=source_system)
  File "/usr/lib/python2.7/site-packages/pymavlink/mavutil.py", line 1235, in mavlink_connection
    return mavudp(device, source_system=source_system, input=input, use_native=use_native)
  File "/usr/lib/python2.7/site-packages/pymavlink/mavutil.py", line 875, in __init__
    self.port.bind((a[0], int(a[1])))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 125] Cannot assign requested address

Looking at what ports are being used by QGC, it seems that it's communicating over UDP with 192.168.1.1:14550... why is it able to connect to Aero while DroneKit can't?

lucasdemarchi, 2017-04-07:
In the default configuration mavlink-router broadcasts the packets to all interfaces. However when the first client responds it switches to unicast to talk to that client only. What's probably happening is that you are connecting with QGC or in any other way with one IP and then trying to connect to it again from another IP.

Try the following and let us know if it works:
- Don't start QGC
- Restart mavlink-router
- Start your dronekit script

It should be able to then talk to it. If you want to permanently add your script you can add one endpoint. In mavlink-router this can now be done on a configuration file in `/etc/mavlink-router/main.conf` (or `/etc/mavlink-router/conf.d/*.conf`).

lucasdemarchi, 2017-04-11:
@majassow any update?

majassow, 2017-04-12:
Without QCG running, I still see the same behavior: connecting using the hosts ip address gives the "Cannot assign requested address" error, and when connecting to the broadcast address 192.168.1.255 I get no heartbeat.

edersondisouza, 2017-04-12:
Hi @majassow,

What's the IP of your computer on Aero WiFi? You should point dronekit to connect to that
IP, not Aero IP. So, if your computer is assigned, for instance, `192.168.1.2`, use that to connect.

For instance, I did a small test here and everything appears ok:

```
>>> from dronekit import connect, VehicleMode
>>> vehicle = connect('192.168.7.1:14550', wait_ready=False)
>>> APM:Copter V3.5-dev (3aa718eb)
```

Note that in my test I've connected to aero via USB-Ethernet, but this shouldn't matter. In this case, `192.168.7.1` is my computer IP (on this network, Aero IP is `192.168.7.2`).

The reason is that, counter-intuitively, when GCS connects to the flight stack via UDP, it doesn't actually connect to the address, but starts _listening_ on the address.

Finally, as a matter of curiosity, mavlink-routerd also serves on TCP port 5760, so you could also connect with dronekit via TCP:

```
>>> from dronekit import connect, VehicleMode
>>> vehicle = connect('tcp:192.168.7.2:5760', wait_ready=False)
>>> APM:Copter V3.5-dev (3aa718eb)
```

See how in this case I've used Aero IP, as this is a 'normal' connection.

majassow, 2017-04-14:
I can connect, but only to the local vehicle.  After some investigation it appears that dronekit does not handle multiple vehicles, which is what I'm attempting.
I will investigate this [dronekit fork](https://github.com/peterbarker/dronekit-python/blob/source-system-filtering/examples/multivehicle/multivehicle.py) and see if it helps, or alternatively go back to using pymavlink directly.

closed 2017-04-14 by majassow

ajeyaajeya, 2019-05-20:
I get dronekit:Link timeout, no heartbeat in last 5seconds.
This is followed by APIException: No Heartbeat in last 30s  aborting.
What am I missing here?