intel-aero/meta-intel-aero #91: Update Yocto to 2.1.2

whbruce, 2017-04-10:
Current OS is built by Yocto 2.1.1. Recommend move to 2.1.2 to get ~200 bug fixes and CVEs. Kernel moves to 4.4.26 but package versions remain the same so user space ABI is unchanged. This should just require changes in repo manifest to contain appropriate poky and meta-openembedded revisions.

lucasdemarchi, 2017-04-10:
This can be done. I think I have the kernel patches rebased which would be the major limitation.

whbruce, 2017-04-10:
Good point on kernel patches. The following didn't apply. 
```
0003-dma-dw-must-initialize-dma-channel.patch 
0012-pmic-intel-port-whiskey-cove-driver.patch 
0019-regulator-whiskey_cove-implements-WhiskeyCove-pmic-V.patch 
0020-pmic-whiskeycove-add-vqmmc-regulator-for-SD-host-vol.patch 
```
I'll update to 2.1.2 but leave kernel at 4.4.3 for the time being.

whbruce, 2017-04-12:
Done. See manifest [PR #4](https://github.com/intel-aero/intel-aero-manifest/pull/8)

closed 2017-04-17 by lucasdemarchi