intel-aero/meta-intel-aero #92: Include Catkin (catkin_make) in image

mindThomas, 2017-04-10:
Is it possible to include Catkin (ROS compiler/toolchain) in the output image such that we can compile ROS nodes/projects on the drone instead of having to cross-compile?

Catkin tools would be very beneficial or at a minimum just the possibility of using "Catkin_make" on the drone will be very usefull.

lucasdemarchi, 2017-04-11:
What's the benefit of compiling on the drone rather than on our machines? For M1 we plan to have a more workable SDK, so I don't see the benefit of having it on the drone. Maybe I'm missing anything?
The image is already huge because of headers, debugs, compiler, etc. Things that should be on an sdk not the final image.

whbruce, 2017-04-11:
Agree with @lucasdemarchi, using SDK (preferably the extensible SDK)  is the best way forward. Ideally the eSDK can be published for others to use.

mindThomas, 2017-04-11:
I do not agree. Being a drone intended for development and research rather than deployment, it should be as easy and quick to get started developing programs for the drone.
Keep in mind that the Intel Aero targets developers not end-users nor end-products.
That being said it is quite funny that you Intel guys apparently share the same opinion, without taking your customers (the developers/researchers) into account.
May I ask you, have you done any market research on the use-cases and customers of the Intel Aero drone?

If you ask the main researchers using ROS they would prefer flying around with Ubuntu if possible, or at least being able to compile and test without any hassle of transferring RPM, packages, binaries etc. before they can rerun their test. Everything costs valuable time.

lucasdemarchi, 2017-04-11:
@mindThomas this is a development repository for meta-intel-aero and whether we/you like it or not, it's Yocto Project. People are welcome to install whatever distro flavor they prefer.

As I said, having a proper SDK (which we don't have right now) should provide the tools and environment to develop it on your host machine and deploy on the drone.

> That being said it is quite funny that you Intel guys apparently share the same opinion, without taking your customers (the developers/researchers) into account.
> ...
> If you ask the main researchers using ROS they would prefer flying around with Ubuntu if possible

Let's keep discussions on this repository technical, shall we? Instead of making such statements, could you answer my question on why you would build everything on the target drone? Why is it better?

mindThomas, 2017-04-11:
Yes, lets indeed keep the discussion technical. We are trying hard to get some results within our own research project while being slightly frustrated of the Intel Aero platform which we expected was more "R&D polished" when released as a development drone.
But I am just happy to be able to contribute to the overall Intel Aero project by testing the latest image, using it in our daily development flow and by then hopefully spotting some bugs along the way.

With some proper documentation of the SDK (that you won't have to compile yourself, but just could be downloaded), it would be tolerable to do the development "offboard". Though before this documentation is to any value in the general workflow it would have to include:
- how to build normal C/C++ project and transfer to the drone, preferrably with one single command line like "make install" which would transfer and install it to a WiFi connected drone
- how to build ROS projects (catkin workspaces) and transfer these to the drone
- how to build normal C/C++ libraries from source to be used in own C/C++ projects and transfer+install this on the drone
- how to build ROS packages/libraries from source and transfer+install these to the drone, eg. RTAB-MAP or ORB-SLAM

> People are welcome to install whatever distro flavor they prefer.
Of course, and due to the nice Intel core Ubuntu also installs flawlessly. Unfortunately Ubuntu does not include the necessary Kernel drivers to activate the RealSense camera, and the regular librealsense kernel drivers for Ubuntu does not work for the Intel Aero drone - we've tried it.
We couldn't get the other camera streams to work either in Ubuntu. So for people who actually wants to use this drone for anything related to obstacle avoidance, mapping etc., using the onboard peripherals, you are stuck with the Yocto project.

icoderaven, 2017-04-17:
As a purported developer end-user, I would like to chime in and mention how frustrating it is to set up the entire paraphernalia to get this foreign build system (yocto) setup. What I would prefer to have is an ubuntu install on which I can natively compile code. As @mindThomas mentions, however, this is not useful if one can't use the RealSense camera (which to me seems odd; there should be a straightforward way to compile a kernel module for the same). As a point of comparison, a lot of the research community's current drones have an ARM based computer (the Odroid series) that runs Ubuntu on top. This makes development easier since the environment on the target platform is familiar and can run the same suite of tools as the primary machines used for development. Further, it is frustrating not to be able to simply use prepackaged (apt-get) binaries to quickly get things up and running.

lfelipe, 2017-04-17:
@icoderaven @mindThomas We hear you, understand and greatly appreciate your feedback, and please believe me when I say that we are bringing this feedback to others here so that we can continuously try to improve on the project.

With that said, unfortunately we cannot provide support for Ubuntu at this specific repository as it is meant for the default image that gets shipped with Intel Aero and that is what we validate with. @mindThomas mentioned that he tried setting up the RealSense camera in a standard Ubuntu install and that it didn't work, could you please reach out to me directly over email so I can follow up on that? (luis dot strano at intel dot com)

With regards to binary packages, the Yocto Project has a few ways to help with that and we're currently trying to get those supported here so you would be able to more easily develop for the platform. We're also trying to do this development and plan for the releases here in Github so there is transparency in the process and you can get a better view of the roadmap. @whbruce is currently investigating catkin to make sure it will fit this development model and we're targeting to get the Yocto Project SDK he mentioned (with documentation on how to use it in our wiki) ready for the next release (we just tagged one release last week, we had one last issue with it which is being fixed by #103 and we'll be doing another spin in the next few days).

icoderaven, 2017-04-17:
Thanks for the feedback! I look forward to being able to use catkin_tools! That will make life much simpler for me!

I understand re: this being a Yocto project, and to be fair, I do see some of the advantages. I've gone through the learning curve and have set up a build system accordingly where I cross compile, build my package, push it to the device, uninstall the previous rpm, and then install my new rpm. I was just trying to add my two cents to answer the previous question posted by a developer.

mindThomas, 2017-04-19:
@lfelipe Please see the following post with our latest issues we ran into when trying to get the RealSense camera working with Ubuntu 14.04 on the Intel Aero: https://github.com/IntelRealSense/librealsense/issues/444#issuecomment-287767248

lucasdemarchi, 2017-04-19:
Humn... Is this the RTF or you are connecting a realsense camera with an OTG cable?

lucasdemarchi, 2017-04-19:
Last time I checked R200 should work on stock kernel after 4.8

mindThomas, 2017-04-19:
@lucasdemarchi It is the RTF drone, hence with the R200 connected over the "special" USB 3.0 cable.

lucasdemarchi, 2017-04-19:
And you are running Ubuntu with which kernel?

mindThomas, 2017-04-19:
We have tried with both Ubuntu 14.04 and then we tried with Ubuntu 16.04 of which we downgraded the kernel 4.4.

lucasdemarchi, 2017-05-04:
@avinash-palleti can you take a look on this?

closed 2017-07-11 by anselmolsm