intel-aero/meta-intel-aero #95: SFTP not working

mindThomas, 2017-04-11:
It is not possible to connect to the drone over SFTP. Only SCP file transfers over SSH is working.
SFTP is tested with both Nautilus and Filezilla.

The OpenSSH config on the drone, located within "/etc/ssh/sshd_config" shows that the sftp server is enabled:
`Subsystem	sftp	/usr/libexec/sftp-server`
Though this binary is nowhere to be found within the drone.

Actually the only parts related to sftp on the drone is as an avahi service, just publishing the sftp service as available even though it is not.

lucasdemarchi, 2017-04-11:
Thanks. I'll take a look on what's needed to enable it.

whbruce, 2017-04-11:
Try `openssh-sftp-server`

mindThomas, 2017-04-11:
@whbruce The command is not found (not included within the Yocto build)
```
root@intel-aero:~/realsense-examples# openssh-sftp-server
-sh: openssh-sftp-server: command not found
```

whbruce, 2017-04-11:
Sorry, I meant that you should add this package to the Yocto build
`CORE_IMAGE_EXTRA_INSTALL += "openssh-sftp-server"`

lucasdemarchi, 2017-04-11:
@whbruce yep, adding it.

PR #96 (2017-04-11): Add openssh-sftp-server
Fix #95
closed 2017-04-11 by lucasdemarchi