intel-aero/meta-intel-aero #111: Release of  OS Image for v1.2

turnersr, 2017-04-18:
Hello,

Is it possible to please upload the relevant OS Image and BIOS Binary/Capsule for v1.2 to the Intel Download Center? 

Thank you! :)

lfelipe, 2017-04-18:
We're doing final round of validation and will be posting it there soon. There have not been any changes made to the BIOS though so you are looking at simply a new iso file.

turnersr, 2017-04-18:
That's awesome! Are you all going to release a complete example of the build using sources, too? 

I've been trying to wrap my head behind the build system. I understand simple examples, but when I have to bring together all the repos and the other dependencies it is really hard to manage all the meta layers. Do you all think you could put everything in one place for each release so folks can simply run bitbake?

lucasdemarchi, 2017-04-18:
@turnersr unless you want to edit the recipes, you don't actually need to build everything.

If you do want to build everything, the workflow is to use repo tool (see https://source.android.com/source/downloading on how to download the tool).

```bash
repo init -u https://github.com/intel-aero/intel-aero-manifest.git
repo sync
cd poky
TEMPLATECONF=meta-intel-aero/conf . oe-init-build-env
bitbake intel-aero-image
```

lucasdemarchi, 2017-04-18:
@turnersr You can also download an image from our CI. This is probably what will become v1.3... if we don't have more bug reports:
http://ec2-52-59-242-235.eu-central-1.compute.amazonaws.com:8080/job/aero-build-master/385/

Click on the iso file there to download

closed 2017-04-18 by lucasdemarchi