intel-aero/meta-intel-aero #115: Migrate customs init scripts to systemd .service

lucasdemarchi, 2017-04-19:
All of our init scripts should install a systemd's .service file if we choose systemd as the init system.  This is a step towards changing the init.

lucasdemarchi, 2017-04-19:
@avinash-palleti can you take a look on this?

whbruce, 2017-04-19:
@avinash-palleti, I can help.

avinash-palleti, 2017-04-19:
@lucasdemarchi will look into it.
@whbruce Thank you :)

avinash-palleti, 2017-04-19:
There are two ways of choosing initialization manager,
http://www.yoctoproject.org/docs/1.6/dev-manual/dev-manual.html#selecting-an-initialization-manager

I think we should go with second one where systemd is used for main image and sysvint for rescue image.

lucasdemarchi, 2017-04-19:
I think I prefer the first one:

```
DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_initscripts = ""
```

Unless there's a drawback I'm not seeing... we don't use a rescue image anyway.

lucasdemarchi, 2017-04-19:
@avinash-palleti remember that we will only be able to do that after the .service files for our custom init have been created. And after we have conducted several tests.

whbruce, 2017-04-19:
I recommend you use appropriate version of Yocto docs. i.e.
http://www.yoctoproject.org/docs/2.1/dev-manual/dev-manual.html#selecting-an-initialization-manager

closed 2017-05-30 by lucasdemarchi