intel-aero/meta-intel-aero #118: Update modem-enable to use to rc-update

avinash-palleti, 2017-04-19:
recipes-support/modem-enable/modem-enable.bb recipe file has below lines,

    install -d ${D}${sysconfdir}/init.d/
    install -d ${D}${sysconfdir}/rc0.d/
    install -d ${D}${sysconfdir}/rc1.d/
    install -d ${D}${sysconfdir}/rc2.d/
    install -d ${D}${sysconfdir}/rc3.d/
    install -d ${D}${sysconfdir}/rc4.d/
    install -d ${D}${sysconfdir}/rc5.d/
    install -d ${D}${sysconfdir}/rc6.d/
    install -m 0755 ${WORKDIR}/modem-enable ${D}${sysconfdir}/init.d/
    ln -sf ../init.d/modem-enable      ${D}${sysconfdir}/rc0.d/K20modem-enable
    ln -sf ../init.d/modem-enable      ${D}${sysconfdir}/rc1.d/K20modem-enable
    ln -sf ../init.d/modem-enable      ${D}${sysconfdir}/rc2.d/S20modem-enable
    ln -sf ../init.d/modem-enable      ${D}${sysconfdir}/rc3.d/S20modem-enable
    ln -sf ../init.d/modem-enable      ${D}${sysconfdir}/rc4.d/S20modem-enable
    ln -sf ../init.d/modem-enable      ${D}${sysconfdir}/rc5.d/S20modem-enable
    ln -sf ../init.d/modem-enable      ${D}${sysconfdir}/rc6.d/k20modem-enable

Need to update this to use rc-update

avinash-palleti, 2017-04-19:
See: http://www.yoctoproject.org/docs/1.8/ref-manual/ref-manual.html#var-INITSCRIPT_PARAMS and http://www.yoctoproject.org/docs/1.8/ref-manual/ref-manual.html#var-INITSCRIPT_NAME and http://www.yoctoproject.org/docs/1.8/ref-manual/ref-manual.html#ref-classes-update-rc.d

whbruce, 2017-04-19:
A couple of things
1. This seems at odds with moving to systemd (#115)
2. Refer to appropriate version of documentation (i.e. 2.1)

avinash-palleti, 2017-04-20:
@lucasdemarchi 
Shall we move this temporarily to use update-rc until we migrate to systemd ?

lucasdemarchi, 2017-04-20:
@whbruce this is a small change that will help with the systemd (init files won't be installed when choosing systemd as the init).  If we can go to systemd without having this as an intermediary step, I'm fine skipping it.

whbruce, 2017-04-20:
Understood. Stick with your plan.

closed 2017-04-26 by lucasdemarchi