intel-aero/meta-intel-aero #119: How do I connect to the local mavlink daemon on the RTF

icoderaven, 2017-04-19:
Hi

I'm trying to connect to the PX4 firmware on the RTF drone using mavros, but I do not know what port to connect to. These are the things I've tried

Stopping mavlink-router using
/etc/init.d/mavlink-routerd.sh stop
and then trying to run mavros on /dev/ttyS1:921600

Trying to run mavros on udp://@:14556

Trying to connect using DroneCode on the aero drone using
from dronekit import connect, VehicleMode
vehicle = connect('tcp:192.168.8.1:5760', wait_ready=False)
vehicle = connect('tcp:127.0.0.1:5760', wait_ready=False)
vehicle = connect('127.0.0.1:14550', wait_ready=False)
vehicle = connect('tcp:192.168.8.1:14550', wait_ready=False)

None of these work. I am, however, able to fly the drone manually just fine, so the firmware is running on there.

Output of get_aero_version.py
BIOS_VERSION = Aero-01.00.12_Prod
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.3.0 (krogoth)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xff

lfelipe, 2017-04-19:
@icoderaven, on your first attempt (killing mavlink-routerd and then trying to connect directly to /dev/ttyS1), the baudrate should not be 921600 but 460800. That is assuming you've updated PX4 recently (since you're using a recent BSP release it looks like you should).

Trying to run mavros on udp://@:14556 -> this should likely be 14550

On the dronekit questions, if they are running on the aero drone itself, likely tcp on 127.0.0.1:14550 should have worked. Did you change the mavlink-router configuration at any point?

icoderaven, 2017-04-19:
I had not updated PX4 recently, but just did (ran aerofc_update.sh /etc/px4-fw/nuttx-aerofc-v1-default.px4)

Trying a baudrate of 460800 (or 921600) after stopping mavlink-routerd gives me an error
serial0: receive: End of file

Running with the udp port 14550 starts the node, but doesn't connect over mavlink.

dronekit connection attempt with tcp:127.0.0.1:14550 gave me a 
socket.error: [Errno 111] Connection refused

No, I didn't change the mavlink-router configuration at any point.

lucasdemarchi, 2017-04-20:
> FPGA_VERSION = 0xff

You also need to update the FPGA (although not related to your problem)

lucasdemarchi, 2017-04-20:
Can you post your `/etc/mavlink-router/main.conf` file?  @edersondisouza can you point to a dronekit sample?

edersondisouza, 2017-04-20:
Hi @icoderaven,

Locally on Aero, you can connect with dronekit using:

```
>>> from dronekit import connect, VehicleMode
>>> vehicle = connect('tcp:127.0.0.1:5760', wait_ready=False)
```

If you want to connect using UDP, then you need to add a new endpoint, for instance, create a file on `/etc/mavlink-router/config.d/local-udp.conf`, containing:

```ini
[UdpEndpoint localhost]
Mode=Normal
Address=127.0.0.1
```

Finally, if nothing works, could you provide us your mavlink-router log file, on `/var/log/mavlink-routerd.conf`?

icoderaven, 2017-04-20:
Oh wait, the dronecode example works on a complete restart, before I source the ROS setup, but it doesn't after sourcing the ROS setup script! In addition, after sourcing the ROS python path, the import command displays an additional statement
Using MAVLink 1.0

Could it be that there are conflicting mavlink libraries between the ROS packages and the system installed ones?

Also, how do I connect using mavros (what I really wanted in the first place)?

rachelruijiayang, 2017-04-23:
icoderaven posted here as well: https://communities.intel.com/thread/113405

I have the same problem; I'm not sure what command to run to get MAVROS to connect to the FCU's MAVLink stream.

Here is my **/etc/init.d/mavlink-routerd.sh**:
```
#! /bin/sh

### BEGIN INIT INFO
# Provides:        MAVLink Router
# Required-Start:  $network
# Required-Stop:   $network
# Default-Start:   2 3 4 5
# Default-Stop:
### END INIT INFO

# Source function library.
. /etc/init.d/functions

# Functions to do individual actions
start(){
	nohup /usr/bin/mavlink-routerd \
		2>/var/log/mavlink-routerd.log \
		>/dev/null &
}
stop(){
	kill `ps | grep -m 1 'mavlink-routerd' | awk '{print $1}'`
}
status(){
	ps | grep -m 1 mavlink-routerd
}

case "$1" in
  start)
	start
	;;
  stop)
	stop
	;;
  restart)
	stop
	start
	;;
  status)
	status
	;;
  *)
	echo "Usage: mavlink-routerd { start | stop | status | restart }" >&2
	exit 1
	;;
esac

exit 0
```

**/etc/mavlink-router/main.conf:**
(In https://github.com/01org/mavlink-router/issues/67, you say that you tested master branch with (basically) the following main.conf and it seems to work fine)
```
[General]
TcpServerPort=5760
ReportStats=false
Log=/var/lib/mavlink-router/

[UartEndpoint uart]
Device = /dev/ttyS1
Baud = 460800

[UdpEndpoint wifi]
Mode = Normal
Address = 192.168.8.255
[... 171 lines cut ...]
[ WARN] [1492122299.779126652]: WP: timeout, retries left 2
[ WARN] [1492122300.780913454]: WP: timeout, retries left 1
[ WARN] [1492122301.781538619]: WP: timeout, retries left 0
[ERROR] [1492122302.782981423]: WP: timed out.
[ WARN] [1492122314.304671724]: TM: Clock skew detected (9223372036.854574203 s). Hard syncing clocks.
[ WARN] [1492122352.504769350]: TM: Clock skew detected (9223372036.854652405 s). Hard syncing clocks.
```
rostopic echo /mavros/state:
```
header: 
  seq: 83
  stamp: 
    secs: 1492122748
    nsecs: 539460283
  frame_id: ''
connected: True
armed: True
guided: False
mode: CMODE(0)
---
```

**It seems like running mavros on 192.168.8.1:14555 comes closest to success? But throughout all of these tests, output of /var/log/mavlink-routerd.log has been "Message to unknown sysid: 1 **

I've been stuck on this for days, and the instructions in the Wiki have been unsuccessful--I very much appreciate your help!

icoderaven, 2017-04-23:
Huh. This looks like it *is* connecting! I'll try it on my RTF tomorrow.
Hmm. Seems like the PX4 firmware is not responding to the MAV_CMD::REQUEST_AUTOPILOT_CAPABILITIES message.
Off the top of my head, two possibilities could exist
a. The connection is only one way, i.e. since we're accessing over UDP, we're somehow not able to send messages up via the mavlinkrouterd
b. The mavlink dialects/versions between mavros and the one in the firmware are different.

I'd place my bet on the former.

lucasdemarchi, 2017-04-23:
> roslaunch /opt/ros/indigo/share/mavros/launch/px4.launch fcu_url:=/dev/ttyS1:115200

2 mistakes here:
- if mavros is talking directly via uart, you shouldn't have mavlink-router running
- baudrate is wrong... After updating the Linux image, did you update the px4 firmware as well? If so baudrate should be 460800

If you want mavros to go through mavlink-router (because you are using mavlink directly for other things, too) then your second approach is correct

lucasdemarchi, 2017-04-23:
> roslaunch /opt/ros/indigo/share/mavros/launch/px4.launch fcu_url:=udp://@192.168.8.1:14555

Dunno exactly what mavros is doing here, but as per your mavlink-router config, it's sending data on port 14550, not 14555

lucasdemarchi, 2017-04-23:
>  But throughout all of these tests, output of /var/log/mavlink-routerd.log has been "Message to unknown sysid: 1"

Disregard this. It's a harmless warning that has been fixed on mavlink-router (and is available on version 1.3.1 of the BSP ir master branch

icoderaven, 2017-04-23:
a. I'm already on the 1.3.1 branch (a336a906566de718eac41a7a0800b3f056cbafd6). get_intel_aero_version.py says
```
BIOS_VERSION = Aero-01.00.12_Prod
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.3.0 (krogoth)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc0
```
b. I get the same output as @rachelruijiayang on port 14555, nothing on 14550. 
c. I could not get mavros to talk over ttyS1 directly even after stopping mavlink-router.
```
[ INFO] [1492657058.177245525]: FCU URL: /dev/ttyS1:460800
[ INFO] [1492657058.177608212]: serial0: device: /dev/ttyS1 @ 460800 bps
[ INFO] [1492657058.177937835]: GCS bridge disabled
[ERROR] [1492657058.178100728]: serial0: receive: End of file
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
```

I don't really care for mavlink router, I just want my tty connection to the autopilot. How do I get that?

mindThomas, 2017-04-25:
@icoderaven Same problem here, also getting `End of file`

lucasdemarchi, 2017-04-25:
> a. I'm already on the 1.3.1 branch (a336a90). get_intel_aero_version.py says

Note you also need to checkout the right manifest. The aero-get-version.py output says you're still on 1.3.0

icoderaven, 2017-04-26:
Ok, pulled a completely new repo, from your tagged manifest. 
`repo init -u https://github.com/intel-aero/intel-aero-manifest.git -m v1.3.1.xml -b refs/tags/v1.3.1`

get_aero_version.py says
```
BIOS_VERSION = Aero-01.00.12_Prod
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.3.1 (krogoth)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc0
```
Still the same issue.
@mindThomas Mind telling me how you got the old mavlink bridge to work?

mindThomas, 2017-04-26:
@icoderaven We have been using the following mavlink_bridge.py script which have now been removed from the image though: https://github.com/intel-aero/meta-intel-aero/blob/80eb214cfde239171d40daba508372176c2168bb/recipes-core/initscripts/initscripts/mavlink_bridge.py
Though with the updated PX4 firmware the baud rate has been changed to 460800 so you would need to change line 10 to reflect this.

This script can then be used to "forward" all Mavlink communication over UDP.
To start the script you execute: `python mavlink_bridge.py <ip>`
Replace `<ip>` with the IP address assigned to your computer connected to the drone (usually 192.168.8.2) if you want to run Mavros on your PC. Otherwise you can use 127.0.0.1

On your computer you can now launch Mavros and connect to the drone by executing:
`roslaunch mavros px4.launch fcu_url:=udp://@192.168.8.1:14550 gcs_url:=udp://127.0.0.1:14558@127.0.0.1:14610`

This line would also allow you to run QGroundControl at the same time - even though you would have to configure QGroundControl to connect to a different port than the default.

AeroChen, 2017-04-26:
Hi, I did something that worked for me at the moment: 
[(https://communities.intel.com/thread/113796)]

Can you guys check if this works for you too?

icoderaven, 2017-04-26:
Eeee! @AeroChen that works for me! I'm a happy panda. Would love this to be fixed properly though. I'm off to collect data!

lucasdemarchi, 2017-04-26:
@AeroChen thanks!

So... it's probably that mavros is not fully configuring the tty, so it doesn't work depending on what was configured that before.

We can do 2 things here:
1) Fix mavros
2) make mavlink-router to put the tty back into raw state when it exits

icoderaven, 2017-04-26:
I would very much prefer option 2. We have a lightweight mavlink C++ ROS client that we use internally as well, and having raw tty access will be great!

mbelluzzo, 2017-04-26:
If booting without mavlink-router on startup I'm able to run roslaunch tapping directly on the serial but:

rostopic echo /mavros/state returns:
```
header: 
  seq: 0
  stamp: 
    secs: 1492665615
    nsecs: 675344188
  frame_id: ''
connected: False
armed: False
guided: False
mode: ''
```
and /mavros/imu/data returns nothing :-(

after startup without mavlink-router running at all: 
*stty -F /dev/ttyS1*
after boot:
```
speed 9600 baud; line = 0;
-brkint -imaxbel
```

with mavros running:
```
speed 460800 baud; line = 0;
min = 1; time = 0;
-brkint -icrnl -imaxbel
-opost
-isig -icanon -iexten -echo
```
after stop mavros:
```
speed 460800 baud; line = 0;
min = 1; time = 0;
-brkint -icrnl -imaxbel
-opost
-isig -icanon -iexten -echo
```
Rebooting with Mavlink-router ON:
```
speed 460800 baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo
```
Stopping mav-router:
```
speed 460800 baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo
```
Running mavros again (wont start)
```
speed 460800 baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo
```
I'm on 1.3.0 with fpga 0xc0 and px4 image that ships with 1.3.0

lucasdemarchi, 2017-04-26:
> I would very much prefer option 2. We have a lightweight mavlink C++ ROS client that we use internally as well, and having raw tty access will be great!

I mean... we should do both.  If your client can't bring the tty back from whatever state it was to a working state, it has the same bug as mavros has.

Making mavlink-router to put the ttyS1 into "raw mode" is only a palliative measure so the bug on other programs is not exercised

lucasdemarchi, 2017-04-26:
@mbelluzzo from your input... what happens if before this step:

> Running mavros again (wont start)

you do:

`stty -F /dev/ttyS1 onlcr`

can you also send the output of stty after mavproxy runs on /dev/ttyS1?

icoderaven, 2017-04-27:
> If your client can't bring the tty back from whatever state it was to a working state, it has the same bug as mavros has.
Fair enough, I was holding off on getting our client to cross compile until I could get mavros to run ^_^ I agree, though.

lucasdemarchi, 2017-04-27:
`stty -F /dev/ttyS1 min 1`

This is the command you need to run before your client/mavros to get it to work (or fix them so they do the right thing). Just from running strace on mavros it looks like it doesn't configure anything if the speed was already the same speed it should be. I.e. it ignores all the other settings it should set.

lucasdemarchi, 2017-04-27:
I'm adding some changes to mavlink-router to try to play nice with other programs, too.

lucasdemarchi, 2017-04-27:
And here is the PR to mavlink-router: https://github.com/01org/mavlink-router/pull/77

mindThomas, 2017-04-27:
I have just tried this PR without any success.

When starting the mavlink-routerd it does not seem to work. I get the following output:
```
(../git/endpoint.cpp:450) Open UART [4] /dev/ttyS1 *
(../git/endpoint.cpp:391) UART [4] speed = 460800
(../git/endpoint.cpp:539) Open UDP [5] 192.168.8.255:14550
(../git/endpoint.cpp:539) Open UDP [6] 192.168.7.255:14550
(../git/mainloop.cpp:504) Open TCP [7] 0.0.0.0:5760 *
(../git/logendpoint.cpp:171) Logging target system_id=1 on 00047-2017-04-18_13-44-01.ulg
(../git/logendpoint.cpp:184) No Log messages received in 5 seconds restarting Log...
```

And when starting Mavros using the following command, there seem to be no connection:
```
roslaunch /opt/ros/indigo/share/mavros/launch/px4.launch fcu_url:=udp://@192.168.8.1:14550
[ INFO] [1492523814.713169232]: Built-in MAVLink package version: 2016.5.20
[ INFO] [1492523814.713203945]: Built-in MAVLink dialect: ardupilotmega
[ INFO] [1492523814.713250234]: MAVROS started. MY ID 1.240, TARGET ID 1.1
```

Output of "stty -F /dev/ttyS1" is:
```
speed 460800 baud; line = 0;
intr = <undef>; quit = <undef>; erase = <undef>; kill = <undef>; eof = <undef>;
start = <undef>; stop = <undef>; susp = <undef>; rprnt = <undef>;
werase = <undef>; lnext = <undef>; discard = <undef>; min = 0; time = 0;
-cread
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke

```

lucasdemarchi, 2017-04-27:
@mindThomas we are talking about connecting mavros directly to the ttyS1, not via mavlink-router. The change in mavlink-router is to keep they ttyS1 in a state that the current version mavros is able to recover.

lucasdemarchi, 2017-04-27:
The fact that when you stopped mavlink-router the tty configuration changed to "speed 1200 baud" and "min = 1" means that PR worked for what it was proposing to do.

mindThomas, 2017-04-27:
You are right and your PR definitely works in this regard.
I can confirm that I, with the help from your script, is able to "deinitialize" the UART port into such a mode that it is possible for Mavros to open it.
```
roslaunch /opt/ros/indigo/share/mavros/launch/px4.launch fcu_url:=/dev/ttyS1:460800
[ INFO] [1492523814.713169232]: Built-in MAVLink package version: 2016.5.20
[ INFO] [1492523814.713203945]: Built-in MAVLink dialect: ardupilotmega
[ INFO] [1492523814.713250234]: MAVROS started. MY ID 1.240, TARGET ID 1.1
```

But if I start the mavlink-routerd I get:
```
root@intel-aero:/usr/bin# mavlink-routerd(../git/endpoint.cpp:450) Open UART [4] /dev/ttyS1 *
(../git/endpoint.cpp:391) UART [4] speed = 460800
(../git/endpoint.cpp:539) Open UDP [5] 192.168.8.255:14550  
(../git/endpoint.cpp:539) Open UDP [6] 192.168.7.255:14550  
(../git/mainloop.cpp:504) Open TCP [7] 0.0.0.0:5760 *
(../git/logendpoint.cpp:171) Logging target system_id=1 on 00207-2017-04-18_14-39-59.ulg
(../git/logendpoint.cpp:184) No Log messages received in 5 seconds restarting Log...
(../git/logendpoint.cpp:171) Logging target system_id=1 on 00208-2017-04-18_14-40-31.ulg
(../git/logendpoint.cpp:184) No Log messages received in 5 seconds restarting Log...
```

And if I then in another terminal try to launch Mavros connecting to the port opened by mavlink-router it does not work:
```
roslaunch mavros px4.launch fcu_url:=udp://@127.0.0.1:14550
[ INFO] [1492526779.302164707]: FCU URL: udp://@127.0.0.1:14550
[ INFO] [1492526779.302693337]: udp0: Bind address: 0.0.0.0:14555
[ INFO] [1492526779.302799041]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1492526779.303034874]: GCS bridge disabled
[ INFO] [1492526779.401643824]: Plugin 3dr_radio loaded and initialized
[ INFO] [1492526779.415097782]: Plugin actuator_control loaded and initialized
[ INFO] [1492526779.420180600]: Plugin altitude loaded and initialized
[ INFO] [1492526779.443015431]: Plugin command loaded and initialized
[ INFO] [1492526779.475136531]: Plugin ftp loaded and initialized
[ INFO] [1492526779.509754714]: Plugin global_position loaded and initialized
[ INFO] [1492526779.513059386]: Plugin hil_controls loaded and initialized
[ INFO] [1492526779.542948913]: Plugin imu_pub loaded and initialized
[ INFO] [1492526779.566225783]: Plugin local_position loaded and initialized
[ INFO] [1492526779.569015600]: Plugin manual_control loaded and initialized
[ INFO] [1492526779.578695796]: Plugin param loaded and initialized
[ INFO] [1492526779.593990314]: Plugin rc_io loaded and initialized
[ INFO] [1492526779.594101593]: Plugin safety_area blacklisted
[ INFO] [1492526779.608515920]: Plugin setpoint_accel loaded and initialized
[ INFO] [1492526779.659892520]: Plugin setpoint_attitude loaded and initialized
[ INFO] [1492526779.685524343]: Plugin setpoint_position loaded and initialized
[ INFO] [1492526779.724713178]: Plugin setpoint_raw loaded and initialized
[ INFO] [1492526779.736922957]: Plugin setpoint_velocity loaded and initialized
[ INFO] [1492526779.763263666]: Plugin sys_status loaded and initialized
[ INFO] [1492526779.777330369]: Plugin sys_time loaded and initialized
[ INFO] [1492526779.782543991]: Plugin vfr_hud loaded and initialized
[ INFO] [1492526779.799106952]: Plugin waypoint loaded and initialized
[ INFO] [1492526779.799240644]: Autostarting mavlink via USB on PX4
[ INFO] [1492526779.799331559]: Built-in SIMD instructions: SSE, SSE2, SSE3, SSSE3, SSE4.1, SSE4.2
[ INFO] [1492526779.799381899]: Built-in MAVLink package version: 2016.5.20
[ INFO] [1492526779.799434713]: Built-in MAVLink dialect: ardupilotmega
[ INFO] [1492526779.799477014]: MAVROS started. MY ID 1.240, TARGET ID 1.1
```
PS. I have tried with both UDP and TCP and localhost or local IP (192.168.8.1).

mindThomas, 2017-05-04:
Having some success with the pull request now updating Mavros: https://github.com/intel-aero/meta-intel-aero/pull/134
I am able to get both Mavros and the Ulog script running on the drone but unfortunately with the same data rate issues as we have seen before.
Is baud rate 460800 still used or can we update this to something faster?

lfelipe, 2017-05-04:
@icoderaven can this issue be closed now?

icoderaven, 2017-05-05:
I haven't tested it yet but I believe that it should be good to go! Thanks for asking!

closed 2017-05-05 by icoderaven

rudja, 2017-05-12:
Hello @icoderaven. 
I am also trying to use mavros with this RTF, but all solutions given in this page didn't succeeded. Did you finally test and succeed in connecting mavros?
Thanks.

koyahatataku, 2019-03-26:
According to this report (file:///C:/Users/koyap_pf6a3gy/Desktop/IntelAero/AutonomousCourse/D1%20-%20Autonomous%20Drone%20Engineer%20-%20Software%20-%20Hello%20World%20Motor%20Control.pdf), I think we don't have to use serial port with mavros(we can use 127.0.0.1:5760 and it is redirected to serial port automatically )

I did like below and succeeded to arm the motors.

roslaunch mavros px4.launch fcu_url:="tcp://:5760@127.0.0.1:5760"
rosservice call /mavros/cmd/arming "value: true"

koyahatataku, 2019-03-26:
According to this report (https://github.com/intel-aero/Documents/blob/master/course/pdf/D1%20-%20Autonomous%20Drone%20Engineer%20-%20Software%20-%20Hello%20World%20Motor%20Control.pdf), I think we don't have to use serial port with mavros(we can use 127.0.0.1:5760 and it is redirected to serial port automatically )

I did like below and succeeded to arm the motors.

roslaunch mavros px4.launch fcu_url:="tcp://:5760@127.0.0.1:5760"
rosservice call /mavros/cmd/arming "value: true"

DRAEYE, 2020-08-09:
Hello, I have come across the same problem nowadays, but the comment to successfully deal with this issue is lost, can I ask you how to deal with this problem?