intel-aero/meta-intel-aero #122: Power down issues in Aero RTF using latest update

turnersr, 2017-04-20:
Hello,

I updated the RTF drone using [CI build](http://ec2-52-59-242-235.eu-central-1.compute.amazonaws.com:8080/job/aero-build-master/lastSuccessfulBuild/artifact/poky/build/tmp/deploy/images/intel-aero/intel-aero-image-intel-aero-20170420164020.iso) . When I go to turn off the drone using the power button nothing happens. 

I expect that I would see some indication that shutdown is occurring, but when I push and hold the power button nothing happens. 

When I type `shutdown now` in the terminal the drone does try to shutoff. However,when drone tries turn off  it keeps printing information related to "serial8250". 

Please the attached image for what happens after I type `shutdown now`
[image]

Thank you.

lucasdemarchi, 2017-04-20:
> I updated the RTF drone using CI build . When I go to turn off the drone using the power button nothing happens.

Were you able to successfully updated or does this happen with the updated version?

turnersr, 2017-04-20:
I successfully updated and even got some code to run that uses the depth camera. I was able to see the results through my attached monitor.

lucasdemarchi, 2017-04-20:
There's nothing I can see here that would handle the power switch. Did you also update the bios? I think this is handled there rather than on the OS level. I'll take a look on it.

lucasdemarchi, 2017-04-20:
Ok, found it.  The  power down request does come via bios, but we are not handling it correctly on v1.3. We were on v1.2. I'll fix it today. Thanks.

PR #123 (2017-04-20): Add back eee-acpi-scripts to handle poweroff
There was a script on eee-acpi-scripts that was handling the shutdown
button. This package got removed when we migrated to meta-intel.
Add it back.

Fix #122
closed 2017-04-20 by lucasdemarchi