intel-aero/meta-intel-aero #126: Build Error: Airmap-1.8-r0

jckchow, 2017-04-24:
I am having issues building the latest Intel Aero Image (version 1.3) from source. I followed the Quick Start Guide and got stuck with the following error with not much information from the log:

```
ERROR: airmap-1.8-r0 do_configure: oe_runmake failed
ERROR: airmap-1.8-r0 do_configure: Function failed: do_configure (log file is located at /home/pi/intel_aero_v1_3/poky/build/tmp/work/corei7-64-poky-linux/airmap/1.8-r0/temp/log.do_configure.3227)
ERROR: Logfile of failure stored in: /home/pi/intel_aero_v1_3/poky/build/tmp/work/corei7-64-poky-linux/airmap/1.8-r0/temp/log.do_configure.3227
Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: Executing shell function do_configure
| NOTE: make clean
| make: *** No rule to make target `clean'.  Stop.
| WARNING: exit code 1 from a shell command.
| ERROR: oe_runmake failed
| ERROR: Function failed: do_configure (log file is located at /home/pi/intel_aero_v1_3/poky/build/tmp/work/corei7-64-poky-linux/airmap/1.8-r0/temp/log.do_configure.3227)
ERROR: Task 3984 (/home/pi/intel_aero_v1_3/poky/meta-intel-aero/recipes-airmap/airmap/airmap_1.8.bb, do_configure) failed with exit code '1'
```

lucasdemarchi, 2017-04-24:
Are you building the BSP on a raspberry pi??? This is not supported.

whbruce, 2017-04-24:
If you had already built an image then updated to 1.3 and re-built in the same workspace this error may have been caused by the change in MACHINE configuration. See PR #98 for details on workarounds.

jckchow, 2017-04-24:
I'm actually building a fresh image on Ubuntu. I think I found the issue, in the Makefile it is missing the "clean: ". I added that manually to the Makefile and it seems to be working now on Ubuntu 14.04.

lucasdemarchi, 2017-05-12:
should be fixed by #142

lucasdemarchi, 2017-05-30:
This is fixed by the updated airmap recipe on master-next branch

closed 2017-05-30 by lucasdemarchi