intel-aero/meta-intel-aero #127: Update Mavros to v0.18.x to get MAVLink 2

mindThomas, 2017-04-25:
Due to the old version of Mavros supplied within the meta-ros layer one can not do any logging of the PX4 data using the Ulogger at the same time as running Mavros on the drone.
This limits any possibility in doing logging of automated drone behaviour through ROS, as this would require both Mavros and Ulog to run at the same time.

I have tried to do some of the changes in meta-ros myself, pulling in the latest MAVLink, but without any luck.
Instead I ran into problems with the message generator not generating proper C++ header files (common.hpp etc.) and after solving that the Python based message generator (mavgen.py) within pymavlink complained about missing the Python library, "future"

Hopefully anyone can point us in any usefull way?
Topic has also been posted here: https://github.com/bmwcarit/meta-ros/issues/489

Kind regards
Thomas Jespersen

mbelluzzo, 2017-04-26:
As far I could check, even with 17.05 we are using mavlink 2.0 in terms of wired-protocol (mavros build properly detect this and sets -DMAVLINK2_COMPAT). Do I don't think the problem is because of the version of mavlink itself, but if is something else (or mavlink 2.0 is not about wired-protocol) can you share the commit on mavros that fix this issue so we can see what we can do if ppl on meta-ros does not update the recipe? Depending of what it is it may be possible to keep a workaround on our recipes.

mhkabir, 2017-04-27:
@mbelluzzo I'm one of the developers of mavros. The current version of mavros + mavlink you ship does not have mavlink 2.0 support. It came in only in 0.18.0 : https://github.com/mavlink/mavros/releases/tag/0.18.0

Therefore our mavros proxy implementation cannot route Mavlink 2.0 packets and the ULog streaming does not work. You would also need to update the version of mavlink to the latest one to make sure that the message definitions are present.

Thanks!

vooon, 2017-04-27:
And more specific: `MAVLINK2_COMPAT` do not enable v2 parsing, it is fix v1 API changes introduced with v2 release. So it allow to use fresh mavlink package with 0.17 version.
Routing of v2 only possible in 0.18 (Kinetic) and up. You should be able to build it on Indigo, but you need mavlink from Kinetic.

lucasdemarchi, 2017-04-27:
@vooon mavlink and pymavlink, right (since now you are using C++11 generator)?

vooon, 2017-04-27:
[mavlink-gbp-release](https://github.com/mavlink/mavlink-gbp-release) ship both. You may place `release/kinetic/mavlink` branch at workspace and do `catkin build`. CMake uses Python 2, so you need to install dependencies first. Most bugs come from `future` (setup.py imports pymavlink, it imports future and pip for some reason reads requirements.txt after setup).

mindThomas, 2017-04-27:
But that would still not allow Mavros to be used together with flight logs being logged with ULogger.

lucasdemarchi, 2017-04-27:
You would need to make ulogger to talk to mavros.

Or when we bump the version and figure out the problem of talking via TCP/UDP  you can put mavlink-router as the only one using ttyS1 since it has embedded ulog support.

vooon, 2017-04-27:
Why? Mavros 0.18 should route v2 without a problem.
Except that mavros itself do not support packet signing, but routing may work (code for signature handling exists, untested).

mindThomas, 2017-04-27:
> Why? Mavros 0.18 should route v2 without a problem.
Exactly. But that would require Mavros 0.18.
The version is 0.17.5 right now, so even by using `release/kinetic/mavlink` of https://github.com/mavlink/mavlink-gbp-release it wouldn't work.

mbelluzzo, 2017-04-27:
I will work on bump mavros version all way to 0.18.7 and update
mavlink-gbp-release to use kinetic. If it happens that we just need to keep
some custom recipes while those are not bumped on meta-ros upstream, we
will do that. If not, I post here whatever is going wrong for ideas on what
can be done.

mindThomas, 2017-04-27:
@mbelluzzo Fantastic. I will await your results and hopefully good news.
Any ETA? Are you working on this problem right now (at the moment) or is it a postponed task?

mbelluzzo, 2017-04-28:
ros-mavlink is already compiling with kinetic version and worked my way to make yocto pick my mavros recipe (0.18.7). A mistaken priority setting was on the way but I fixed. Now I need to properly update my 0.18.7 recipes with the correct checksums etc and hope for the best during compiling, but this will tomorrows task cause I have to leave.

mbelluzzo, 2017-04-28:
Is working, going to send pull requests in a few :-)

rostopic echo /mavros/imu/data
```
---
header: 
  seq: 1186
  stamp: 
    secs: 1493406149
    nsecs: 560807872
  frame_id: fcu
orientation: 
  x: 0.0197534165702
  y: 0.000656454287714
  z: -0.720316937721
  w: -0.693363551312
orientation_covariance: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
angular_velocity: 
  x: -0.000771123450249
  y: -0.00155919569079
  z: -2.85540590994e-05
angular_velocity_covariance: [1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07]
linear_acceleration: 
  x: -0.112540595233
  y: -0.150820448995
  z: 9.93523979187
linear_acceleration_covariance: [8.9999999999999985e-08, 0.0, 0.0, 0.0, 8.9999999999999985e-08, 0.0, 0.0, 0.0, 8.9999999999999985e-08]
 ```

mhkabir, 2017-04-28:
Looks good :)

mindThomas, 2017-04-28:
Can you please try this with the Syslogger functionality (Ulog) by launching/using the GCS bridge functionality of Mavros.
After enabling the bridge you will have to use/test the Python script: mavlink_ulog_streaming.py
Please see: https://github.com/mavlink/mavros/issues/663

vooon, 2017-04-29:
FYU: I plan to release mavlink and mavros 0.19 at 5/5/2017. (i'm bored with high patch number, signing moving to 0.20).

lucasdemarchi, 2017-04-29:
@voon thanks for the heads up.

mbelluzzo, 2017-05-01:
@mindThomas 

This was the farthest I could go with the Ulog script:
 ```
Output file name: ./2017-05-01_10-53-16.ulg
Connecting to MAVLINK...
Connecting with MAVLink to udp:192.168.8.2:14444 ...
HEARTBEAT OK

Locked serial device

Starting log...
sending heartbeat
sending heartbeat
sending heartbeat
sending heartbeat
Traceback (most recent call last):
  File "./mavlink_ulog_streaming.py", line 267, in <module>
    main()
  File "./mavlink_ulog_streaming.py", line 256, in main
    mav_log_streaming.read_messages()
  File "./mavlink_ulog_streaming.py", line 101, in read_messages
    raise Exception('Start timed out. Is the logger running in MAVLink mode?')
Exception: Start timed out. Is the logger running in MAVLink mode?
 ```
Other than that, was able to connect QGC using mavros routing through gcs_url:=@<my ip>:14444

mavros gcs_briged didn't work here complaining about 'Invalid roslaunch XML syntax'.

mindThomas, 2017-05-01:
@mbelluzzo Wait... Are you connecting/starting Ulog on your computer or on the drone?
At least the IP address 192.168.8.2 does not seem correct as the drone have 192.168.8.1, or if running this Ulog script on the drone it is better to just use 127.0.0.1

mbelluzzo, 2017-05-01:
On my computer because mavutil on the drone is too old to run the script
properly (even with the bump)

mhkabir, 2017-05-01:
Can't you update `mavutil` too? Without being able to run logging on the drone, it really breaks user experience.

lucasdemarchi, 2017-05-02:
We should add instructions to make mavros connect to mavlink-router besides the instructions to connect mavros directly. mavlink-router has logging capabilities embedded.

One could just install the newest pymavlink with pip and have it working without messing with the base system (I actually rsync/scp my site-packages dir and export PATH accordingly)

mindThomas, 2017-05-04:
With the image today I can confirm that the merged pull requested are indeed working with an updated version of Mavros included.
Using pip to update pymavlink allowed me to connect and run Ulog, just as desired. Thanks.

Unfortunately the actual bottleneck problem has now surfaced - our rate and dropout issues with Ulog was not due to WiFi limitations: https://github.com/intel-aero/meta-intel-aero/issues/138

closed 2017-05-04 by mindThomas

zxp771, 2018-05-17:
@mbelluzzo 
hi
how do you get the imu data with ros?
Do these data is synchronized with the camera data?
Is it available for using ros to get the gps data?
Thanks for your help.

mbelluzzo, 2018-05-22:
@zxp771 You can find all the info about mavros here: http://wiki.ros.org/mavros#Overview