intel-aero/meta-intel-aero #133: Can't update px4 firmware with 1.3.1

majassow, 2017-04-27:
Upgraded to 1.3.1.
If I run aerofc-update.sh nuttx-aerofc-v1-default.px4 I get the following output:

BIOS_VERSION = Aero-01.00.12A_Prod
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.3.1 (krogoth)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xff

Updating firmware on AeroFC
==========================================================================================================
WARNING: You should uninstall ModemManager as it conflicts with any non-modem serial device (like Pixhawk)
==========================================================================================================
Loaded firmware for 41,0, size: 888356 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyS1 with baudrate=1500000...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyS1 with baudrate=460800...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyS1 with baudrate=115200...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyS1 with baudrate=1500000...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyS1 with baudrate=460800...

I've disabled ModemManager, and I've also disabled the mavlink_routerd.
I've also tried removing and re-inserting the px4 module to see if that helps.

lucasdemarchi, 2017-04-27:
First of all you need to update the FPGA. It shouldn't cause this problem, but it should be updated nonetheless.. after that you will see the FPGA version set to 0xc.

Is the flight stack running on the flight controller? What is its version? Can you communicate with it using mavlink_shell?

If you have updated before to a broken version you can force to enter in bootloader mode:
1) flash the recovery FPGA version (see in /etc/FPGA
2) remove power from the rtf
3) power it up again
4) you will see the flight controller blinks 6 of the leds on the 7-segment "display". This means it stopped on bootloader and is waiting for new firmware
5) flash the flight controller with the .px4 file
6) flash the rtf FPGA again
7) remove power and power it up again

lucasdemarchi, 2017-04-27:
Don't bother with removing  modemmanager. That warning is bogus

majassow, 2017-04-27:
Thanks! 
OK.  Did that.  px4 is now updated.
I have recalibrated Accel, Gyro, but compass calibration fails.
It seems to be working fine, detecting each orientation, but after the last orientation it immediately complains about a sensor failure: mag0.

Here is the output from mavlink-shell:
nsh> sensors
INFO  [sensors] usage: sensors {start|stop|status}
nsh> sensors status
INFO  [sensors] gyro status:
INFO  [lib__ecl] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [lib__ecl] sensor #0, prio: 99, state: OK
INFO  [lib__ecl]        val:   0.0002, lp:   0.0003 mean dev:   0.0000 RMS:   0.0032 conf:   1.0000
INFO  [lib__ecl]        val:  -0.0048, lp:  -0.0026 mean dev:   0.0000 RMS:   0.0042 conf:   1.0000
INFO  [lib__ecl]        val:  -0.0008, lp:  -0.0005 mean dev:   0.0000 RMS:   0.0022 conf:   1.0000
INFO  [sensors] accel status:
INFO  [lib__ecl] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [lib__ecl] sensor #0, prio: 99, state: OK
INFO  [lib__ecl]        val:  -0.4044, lp:  -0.3906 mean dev:  -0.0002 RMS:   0.0599 conf:   1.0000
INFO  [lib__ecl]        val:   1.7013, lp:   1.7300 mean dev:  -0.0001 RMS:   0.0733 conf:   1.0000
INFO  [lib__ecl]        val:  -9.6612, lp:  -9.6458 mean dev:   0.0000 RMS:   0.0593 conf:   1.0000
INFO  [sensors] mag status:
INFO  [lib__ecl] validator: best: -1, prev best: -1, failsafe: NO (0 events)
INFO  [sensors] baro status:
INFO  [lib__ecl] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [lib__ecl] sensor #0, prio: 75, state: OK
INFO  [lib__ecl]        val:  62.6189, lp:  61.8740 mean dev:  -0.0064 RMS:   0.4744 conf:   1.0000
INFO  [lib__ecl]        val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   1.0000
INFO  [lib__ecl]        val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   1.0000
INFO  [sensors] Temperature Compensation:
INFO  [sensors]  gyro: enabled: 0
INFO  [sensors]  accel: enabled: 0
INFO  [sensors]  baro: enabled: 0
nsh>

output from aero-get-version.py

BIOS_VERSION = Aero-01.00.12A_Prod
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.3.1 (krogoth)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc0

lucasdemarchi, 2017-04-27:
```
INFO [sensors] mag status:
INFO [lib__ecl] validator: best: -1, prev best: -1, failsafe: NO (0 events)
```

Do you have the external mag connected?  It looks like it's not there.  I think this issue should be moved over to PX4 repository and we continue discussing there.

lucasdemarchi, 2017-05-03:
I think we talked on gitter and you solved the problem... ? Can we close this issue?

closed 2017-05-04 by lucasdemarchi

jsfurtad, 2017-11-13:
Hi majassow,

Can you tell me how you disabled the ModemManager? I can't get commands like apt-get to work in the Yocto terminal.

zehortigoza, 2017-11-13:
@jsfurtad please open a new issue also please explain I little bit more.