intel-aero/meta-intel-aero #137: Stream VGA camera video over Wi-Fi

vankamm, 2017-05-03:
Trying to walk through example "WiFi streaming using QGroundControl".  The command to launch gstreamer isn't working for me.  I notice it says to use 'video13', but that device is not listed in my dev folder.  I only have video0-9.  I actually just tried all of them with no luck.  What is the VGA camera mapped to?

vankamm, 2017-05-04:
A quick follow-up to my question...  I now realize that example was for using the RealSense camera.  I plugged that in and the example works great.  I am now seeing other posts related to getting the VGA camera working and will take a look at those.  If you have any other suggestions for getting the VGA camera to work please let me know.

mindThomas, 2017-05-15:
So far we have also only had success with the RealSense camera - both depth and color though.
We would like to know about how to use and interact with the two VGA cameras, especially for using them with ROS.

zehortigoza, 2017-08-07:
The down-face camera is available in /dev/video2 id=1
More info and link to examples in here: https://github.com/intel-aero/meta-intel-aero/issues/223

zehortigoza, 2018-06-05:
Latest release have the VGA stream in camera-daemon-streaming

closed 2018-06-05 by zehortigoza

malfonsoNeoris, 2019-03-06:
> A quick follow-up to my question... I now realize that example was for using the RealSense camera. I plugged that in and the example works great. I am now seeing other posts related to getting the [...]

Hi. i am starting with the intel aero RTF... already installed ubuntu, and i have acces to the R200 cameras... but still no luck on getting the down-face and front-face camera.
could yo provide some info? my main language for prototyping is python but, could also be c++.

Also... i am having problem with the compass, not being detected by QGroundControl, or MissionPlanner ( i have installed ardupilot). Do you have any of this issues?

thanks