intel-aero/meta-intel-aero #141: Dronekit can't connect after sourcing /opt/ros/indigo/setup.sh

jckchow, 2017-05-11:
I can connect to the Intel Aero RTF drone using Dronekit without any problems. But if I try to run the following command first:  

_export PYTHONPATH=$ROS_ROOT/lib/python2.7/site-packages:$PYTHONPATH_

Then that same connect function raise APIException('Timeout in initializing connection"). I'm trying to connect to the vehicle on tcp:127.0.0.1:5760

Any idea how I can setup ROS and use Dronekit at the same time on the Intel Aero?

mindThomas, 2017-05-15:
Without having tested nor confirmed if this would work, I suggest that you investigate and try the gcs bridge feature of Mavros. After Mavros has been updated on the drone we can now launch Mavros on the drone, using it as intended.

1. To launch Mavros on the drone you first need to stop the mavlink-router which is started at power-up:
`sh /etc/init.d/mavlink-routerd.sh stop`

2. As mavlink-router does not seem to reset the state of the serial port correctly we will do this manually:
`stty -F /dev/ttyS1 min 1`

3. Now you can launch Mavros on the drone. I would suggest using a screen but you can also just start the process in the background by calling:
`roslaunch mavros px4.launch fcu_url:=/dev/ttyS1:460800 gcs_url:=udp://@127.0.0.1`

4. The above command also starts a local gcs bridge over UDP, which can also be replaced by tcp eg. for your usecase with Dronekit. Otherwise you can also bridge this local gcs to a connect computer running QGroundControl etc. This is done by (replace x with the IP of your computer):
`rosrun mavros gcs_bridge _gcs_url:='udp://:14560@192.168.8.x:14561'`

5. After doing so you need to configure QGroundControl and point to 192.168.8.1:14560 and listen on port 14561.

lucasdemarchi, 2017-06-02:
After some debug we discovered a way to run with mavlink-router as well.  The problem is that mavros uses sysid=1 that's the same in use for the autopilot. This means mavlink-router skips the endpoint of the flight stack to avoid sending the message in a loop.  mavros uses compid to differentiate itself, which is not correctly handled.  I'm running mavros here wth:

```
rosrun mavros mavros_node _fcu_url:=tcp://127.0.0.1:5760 _system_id:=2
```

This tells mavros to use another system_id. No need to create the bridge for GCS since this is already handled by mavlink-router, as well as saving the flight stack log.  If you prefer to use udp, you can use a similar approach:

- Add a mavros.conf in `/etc/mavlink-router/conf.d/` with:
```
[Endpoint mavros]
Address = 127.0.0.1
Port = 14650  # need to be different from 14550 that is used by the "udp endpoint" by default
```
- Start mavros:
```
rosrun mavros mavros_node _fcu_url:=udp://127.0.0.1:14650@ _system_id:=2
```

lucasdemarchi, 2017-06-02:
A fix to mavlink-router is being worked on as well.

closed 2017-06-02 by lucasdemarchi