intel-aero/meta-intel-aero #143: Connection to RTF with mavros running on my computer

rudja, 2017-05-12:
Hello everybody,

I was able to perfectly command my RTF drone through mavros running on my computer, but after the last update (1.3.1), I can't... I saw that I was not alone referring to #119, but after testing all solutions on this page, impossible to connect with the drone. 
There is a new way to do that or I need to wait according this issue #134 ?
I am on ubuntu 14.04, and my drone with the last update. 

Thanks

mindThomas, 2017-05-15:
I think several issues are still present with the UART configuration and Mavlink-routerd. At least we only have success with Mavlink-routerd with QGroundControl.
Let me present the steps that we are currently succefully using with the latest build of the image, although using the outdated and previously removed mavlink_bridge script:

1. After powering up the drone you need to stop mavlink-routerd. Connect to the WiFi of your drone and SSH into it:
`sh /etc/init.d/mavlink-routerd.sh stop`

2. As mavlink-routerd does not reset the serial port state properly we do this manually:
`stty -F /dev/ttyS1 min 1`

3. Download and transfer the mavlink_bridge python script. Execute the following in a regular shell on your computer connect to the drone over WiFi:
```
wget https://gist.githubusercontent.com/mindThomas/51f10f9f2c1d9aa2650adce55bc001f1/raw/5acd50c38c8dd53d3eaa001c759ab5ed668aadf6/mavlink_bridge.py
scp mavlink_bridge.py root@192.168.8.1:mavlink_bridge.py
```

4. Start the mavlink_bridge pointing it towards the IP address of your computer (replace x) running Mavros:
`python mavlink_bridge.py 192.168.8.x &`

5. Start Mavros on your computer. Please notice that your Mavros should be version 0.18 or newer.
`roslaunch mavros px4.launch fcu_url:=udp://@192.168.8.1:14550 gcs_url:=udp://127.0.0.1:14558@127.0.0.1:14610`

6. You should now at least see some heartbeat messages in the terminal on your computer. You can also check the connection by verifying the Connected boolean within the /mavros/state topic:
`rostopic echo /mavros/state`

rudja, 2017-05-16:
Thx for your answer @mindThomas . 
I tried today your solution and It is successful! My problem was the version of Mavros always in 0.17...
Have a nice day!

closed 2017-05-16 by lucasdemarchi

rudja, 2017-05-30:
@mindThomas Hello, 
I have a question for you. Did you success in adding some sensors (Lidar, GPS...) through the FPGA part of this drone? 
I am asking that because after my first test, I have some imprecision.
I am now trying to add some sensors to improve his performence. 

Thanks.

mindThomas, 2017-05-30:
We have not tried adding any other sensors through the FPGA part.
Our project has mainly been focusing on using the on-board RGB-D camera and the PX4 flight controller within a distributed ROS environment.
We succeeded in developing, configuring and implementing an indoor position controller solution for the Intel Aero, using our SLAM-based position estimator fed with measurements from the RGB-D camera.

rudja, 2017-05-31:
Thanks for your reply. 
Great job! It will be my next step :). If you know someone doing my step, it will be nice if you could advert me! 
Have a nice day @mindThomas!

zxp771, 2018-05-22:
Hi @mindThomas 
I saw you were doing the slam research on the intel Aero RTF.
I also want do the research on slam like visual-inertial algorithm.
I have some question I hope you can help me.
1.Could I change the imu operating frequency by myself.I record the imu/data it seems like its operating frequency is 50hz. I'm wondering if the imu can reach 100 or 200hz. because for most visual-inertial algorithms the data should be collect in this frequency.
2.how can I do the extrinsic parameter calibration(camera between imu)?
3.Does the imu and camera data I get from ros should in synchronization condition?I use rosbag fuction to record these data.

Thanks