intel-aero/meta-intel-aero #144: Spare parts with shipping to EU - dramatic crash with broken arm

mindThomas, 2017-05-15:
Hi.
While both the development progress and the overall project with the Intel Aero drone was going very well we unfortunately had quite a dramatic crash today.
Using just the PX4 controller, flying indoor in a Vicon (Mocap) equipped room with the settings enabled properly such that we were able to do Position hold and controlled movement with the drone. This has been working fine for weeks and today we simply had to grab some camera measurements while using the stabilized position.
After taking off the yaw axis was moving slightly and the position lock was not as good as we have previously seen it. All of a sudden though the drone sky rockets into the ceiling where it sits for a few seconds while scraping the ceiling with the propellers, eventually causing them to break. With four broken propellers the drone hammers into the ground unfortunately landing straight on one of the arms.

Luckily it seemed like only one arm, and just the carbon fiber arm, has been damaged. The motor seem to be working resonably (no noisy/scratching sound) and the rest of the frame and housing is intact.

**This leads me to the question of buying a replacement carbon fiber arm shipped to Denmark**
It looks like the following item is what we need but it seems like only shipping within the US is possible.
http://shop.yuneecusa.com/Aero-Drone-Carbon-Fiber-Motor-Arm

This is of course very unfortunate for the project and might end up having dreadful consequences to our results as the project is set for a deadline within 2 weeks.

Let me know which options we have?

Regards Thomas

lfelipe, 2017-05-15:
Let me take a look at this and get back to you.

lfelipe, 2017-05-30:
Shipping of spare parts is being handled by Yuneec, we're still waiting on them to reply and we'll escalate to make sure that they are shipping to more countries.

mindThomas, 2017-05-30:
Ha, too late. Obviously the customer service level from Intel (your side) is really lacking and quite slow.
If you want the community to work with your equipment and move to your equipment you need to step up and support the community as well.
So many other nice drone frames, DIY assemblies or Ready to Fly kits are already available on the market with great community support, replacement parts and several webshops.

I must admit that even though we managed to get our project and system working with a fully-functional RGB-S SLAM implementation for indoor flight on the Intel Aero, it has not been an easy task. The Intel Aero hardware is not really designed for the drone community and hackers - not to mention that the mechanical structure shows a lack of proper knowledge and experience of designing drone frames.

The flight controller, being embedded on to the Intel Aero board, is both vibrating and totally undampened. This causes critical vibrations of the accelerometer making it difficult (almost impossible) to attain any good flights if the accelerometer data is used.
Why is it that the Intel Aero board is not even screwed into the frame or at least taped down or with other foam held down so it can't vibrate inside the plastic enclosure?

In any case we had to solve the broken arm ourselves, but thanks to the craftsman within our lab we managed to get the drone flying again.

lfelipe, 2017-07-12:
@mindThomas I understand your concerns there. I'm glad to hear you got the drone flying again, and I've forwarded your feedback to the relevant people here. I'm closing this as this is more of an issue to the customer forums than for this repository (which focuses directly on the software side of the platform), and I've seen you also brought the issue there. Thanks for the honest feedback, and look forward to more information about your project and what you have been trying to accomplish there. Hopefully some of the changes we have made for the 1.4 release will make development easier for you as well.

closed 2017-07-12 by lfelipe