intel-aero/meta-intel-aero #145: build_version.sh: move to a first-boot service

lucasdemarchi, 2017-05-16:
The script broke on master-next which means aero-get-version is broken.

avinash-palleti, 2017-05-19:
@lucasdemarchi Is this issue still there? I am able to get OS_Version and AIRMAP_Version in master-next build. 
BIOS and FPGA versions are coming as unknown as I am on Joule.

lucasdemarchi, 2017-05-19:
Issue is still there. If you can get the version from aero-get-version.py it's probably because systemd is handling it as a compat sysvinit script

padmashree9110, 2017-05-25:
@lucasdemarchi I made below changes. It is working now. Initially i tried doing changes in initscripts.bbappend only, but inherit systemd there was causing other issues. Seems to be bug with Yocto. So i created new recipe. There is one more patch which adds this recipe in image.bb file.

https://github.com/padmashree9110/meta-intel-aero-base/commit/702e9ca9ad23f9293ad8482e58514979af51f56d

padmashree9110, 2017-05-25:
@lucasdemarchi if you are fine with changes, i will submit PR

lucasdemarchi, 2017-05-25:
Submit a PR, we can review there. Thanks

avinash-palleti, 2017-05-31:
@lucasdemarchi As you suggested using dmidecode utility to get version will be cleaner. We will add that utility into image and update aero-get-version.py accordingly.

padmashree9110, 2017-05-31:
https://github.com/intel-aero/meta-intel-aero/pull/161
https://github.com/intel-aero/meta-intel-aero-base/pull/4

closed 2017-06-06 by lucasdemarchi