intel-aero/meta-intel-aero #146: boot complete notification with RGB led is broken

lucasdemarchi, 2017-05-16:


lucasdemarchi, 2017-05-16:
need to add proper systemd service

avinash-palleti, 2017-06-01:
init_start.sh and init_complete.sh which are under meta-intel-aero-base/recipes-core/initscripts/ have to be started by a systemd service right ?

vidhyagithub, 2017-06-01:
Placing above scripts (init_start.sh & init_complete.sh) across the desired run levels of INIT (/etc/rcx.d) might solve this issue. Also it has to be noted that reboot&update script (meta-intel-aero-base/recipes-support/scripts/files/aero-reboot-update.py) too controls LED pattern during reboot&update.

avinash-palleti, 2017-06-01:
@vidhyagithub INIT scripts will not work as we moved to systemd

Rjasuja, 2017-06-05:
@lucasdemarchi i have added the service files for init_start and init_complete in this patch : https://github.com/Rjasuja/meta-intel-aero-base/commit/e9de27d1dc23f6d8a1f348bf3b308e0cc9a16f01
Please review

closed 2017-06-23 by lucasdemarchi