intel-aero/meta-intel-aero #147: Start using systemd-boot as bootloader

lucasdemarchi, 2017-05-16:
- Simpler configuration files
- Integrate well with systemd
- Allow easy "boot to target just on next boot", without having to package extra tools
- Integrate well with custom-kernel script

lucasdemarchi, 2017-05-16:
Currently this is triggering this bug: https://bugzilla.yoctoproject.org/show_bug.cgi?id=11503

avinash-palleti, 2017-05-17:
I did below changes in master-next,
`diff --git a/conf/machine/intel-aero.conf b/conf/machine/intel-aero.conf
index a921cd3..1531571 100644
--- a/conf/machine/intel-aero.conf
+++ b/conf/machine/intel-aero.conf
@@ -23,8 +23,10 @@ MACHINE_EXTRA_RRECOMMENDS += "efibootmgr"
KERNEL_MODULE_AUTOLOAD += "spi_imu spi_fpga spi_can"

# Grub config
-EFI_PROVIDER = "grub-efi"
+#EFI_PROVIDER = "grub-efi"
+EFI_PROVIDER = "systemd-boot"
APPEND += "console=ttyS0,115200n8 console=tty1"
GRUB_TIMEOUT = "3"
LABELS_LIVE = "install boot"
+LABELS = "install boot"
SERIAL_CONSOLES = "115200;ttyS0"
`
When I use dd command to copy image to usb, it is not showing boot and install options while booting from USB.
When I copy image to windows PC and use Rufus utility to copy to USB, it is showing boot and install options and it is booting to shell as well if I choose boot option.

I checked in Rufus documentation that this utility will do “partition and format the USB in a way that Windows can always understand (and using the whole capacity of the drive), and then copy each individual file and directory from the ISO image onto the newly created file system.”
[image]
[image]

lucasdemarchi, 2017-05-17:
( @avinash-palleti tip to get correct formatting on github: use "```" for code and bash snippets, so it uses monospace font )

So... Yes, I got this change you did a quick test. Note there's a warning about using LABELS. You should rather use LABELS_LIVE. This is the patch I applied to test:
https://gist.github.com/lucasdemarchi/dfa86950e55ffcbabc013a47106759b0

I also added a quick patch on `poky/meta` to copy the configuration files over to the second partition... not sure why it didn't work... this needs to be investigated.
https://gist.github.com/lucasdemarchi/b5fb815c638796bf19b707b0142c82c6

The reason it works with Rufus is that it is actually redoing the image. It opens the .iso, format the usb drive, create a partition, the bootloader and copies the content over there. We need to make this work on the image we generate.

avinash-palleti, 2017-06-02:
@lucasdemarchi After adding loader into EFI partition i am able to see boot entries on Aero and Joule. On Joule it is booting fine but on Aero it is giving blank screen after selecting either of the boot entries (boot or install). No console logs are coming, How to get serial logs on Aero ?

lucasdemarchi, 2017-06-02:
It should show on the monitor. What you can also do: 
- if it's an assembled rtf, flash the FPGA with the version of the compute-board-only (see /etc/FPGA)
- If it's the board with aerofc connected, you can simply disconnect it.

Then edit the kernel command line on the bootloader and replace ttyS0 with ttyS1. This should allow you to use the serial pins on the power connector to see the logs.

lucasdemarchi, 2017-06-02:
It may be kernel options that we don't enable. Please check the systemd-boot requirements wrt kernel config options and check if we are enabling them.

avinash-palleti, 2017-06-12:
It might be BIOS issue as same build is booting fine on Joule board.
Currently i am using Aero-01.00.11 bios, Is there any latest version to this with which i can check.